introlab/rtabmap #249: Can a sparse cloud be an input to obstacles_detection nodelet?

seiya0412, 2017-12-15:
Hi.

While I was trying to use obstacles_detection nodelet on ROS, I received the following exception and the nodelet has died.

```shell
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2017-12-15 21:22:02.179) util3d_filtering.cpp:146::voxelize() Condition ((cloud->is_dense && cloud->size()) || (!cloud->is_dense && indices->size())) not met! [Cloud size=73295 indices=0 is_dense=false]
[obstacles_detection-7] process has died [pid 4022, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet standalone rtabmap_ros/obstacles_detection cloud:=stereo/points2_sparse __name:=obstacles_detection __log:=/home/xxxx/.ros/log/0b0fda7e-e13a-11e7-a471-b808cf52703a/obstacles_detection-7.log].
```

I looked through rtabmap_ros and rtabmap, and I found that it seems like `pcl::IndicesPtr(new std::vector<int>)` is thrown to `OccupancyGrid::segmentCloud()` and `util3d::voxelize()` as an input indices.
In this case, if I understand correctly, `indices->size()` in line 146 of util3d_filtering.cpp is always ZERO and UASERT_MSG is always called.
Does this mean I cannot throw a sparse cloud as an input to obstacles_detection nodelet?
Or, am I just doing something wrong/stupid?

Thank you very much for your help.
Best regards,

Seiya 

--------------

(https://github.com/introlab/rtabmap_ros/blob/b8f8c56d1d30b531d4685dac31d64a8186eea305/src/nodelets/obstacles_detection.cpp#L284-L291)
```C++
pcl::PointCloud<pcl::PointXYZ>::Ptr cloud = grid_.segmentCloud<pcl::PointXYZ>(
		inputCloud,
		pcl::IndicesPtr(new std::vector<int>),
		pose,
		cv::Point3f(localTransform.x(), localTransform.y(), localTransform.z()),
		ground,
		obstacles,
		&flatObstacles);
```

https://github.com/introlab/rtabmap/blob/c21d478f5dafcd4589bbfe89e313f04ee6c74a08/corelib/include/rtabmap/core/impl/OccupancyGrid.hpp#L38-L50

https://github.com/introlab/rtabmap/blob/c21d478f5dafcd4589bbfe89e313f04ee6c74a08/corelib/src/util3d_filtering.cpp#L139-L146

seiya0412, 2017-12-15:
I changed https://github.com/introlab/rtabmap_ros/blob/b8f8c56d1d30b531d4685dac31d64a8186eea305/src/nodelets/obstacles_detection.cpp#L284-L291 to the following lines of code.
I'm not sure if it's correct way to "fix", but at least the nodelet now works and I got obstacles/ground as I desired. 
I would be grateful if you could comment/advice if you have any concern about this change. Thank you.

(BTW, I found that I should post this issue to rtabmap_ros repo instead of rtabmap repo. Sorry.)

```C++
pcl::IndicesPtr inputIndices(new std::vector<int>);
inputIndices->resize(inputCloud->size());
for(unsigned int ii = 0; ii < inputCloud->size(); ii++)
{
	inputIndices->at(ii) = ii;
}
pcl::PointCloud<pcl::PointXYZ>::Ptr cloud = grid_.segmentCloud<pcl::PointXYZ>(
	inputCloud,
	inputIndices,
	pose,
	cv::Point3f(localTransform.x(), localTransform.y(), localTransform.z()),
	ground,
	obstacles,
	&flatObstacles);
```

matlabbe, 2017-12-15:
In [devel branch](https://github.com/introlab/rtabmap/blob/devel/corelib/src/util3d_filtering.cpp#L186-L193) of rtabmap, the assert has been removed from voxelize functions. However the conditions are still there in order to filter or not the cloud. The real problem here is that indices are empty while the cloud is not dense. The field [is_dense](http://docs.ros.org/api/sensor_msgs/html/msg/PointCloud2.html) of the input cloud may be false as it should be true.

Is it a cloud coming from point_cloud_xyz nodelet? If so, what are the parameters of this nodelet?

seiya0412, 2017-12-18:
> Is it a cloud coming from point_cloud_xyz nodelet?

No, the cloud comes from OpenCV's [omni::stereoReconstruct()](https://docs.opencv.org/trunk/d3/ddc/group__ccalib.html#ga907c0cb0b0ec8896f1e60510b546e63a).
Since omni directional camera's projection model is different from pinhole/fisheye model, point_could_xyz nodelet (and `rtabmap::util3d::cloudFromDisparity`) is not suitable to use.

The output cloud from `omni::stereoReconstruct()` is a 6-channel float32 vector (x, y, z, r, g, b). So I convert it to `sensor_msgs::PointCloud` first, then use `sensor_msgs::convertPointCloudToPointCloud2()` to generate PointCloud2 message which is fed to obstacles_detection nodelet.

I guess there would be some mistakes of me in the conversion sequence. I'm going to double check characteristics of point cloud variables.