introlab/rtabmap #250: Google kills its Tango augmented reality platform, shifting focus to ARCore

thibaultmol, 2017-12-16:
https://techcrunch.com/2017/12/15/google-kills-its-tango-augmented-reality-platform-shifting-focus-to-arcore/
https://twitter.com/projecttango/status/941730801791549440

So what does this mean for rtabmap? I hope it keeps getting support for it, as long as it's doable? (Recently bought a used Lenovo Phab pro 2 for mapping some my real estate)

matlabbe, 2017-12-17:
Hi, as long as the current tango core works, rtabmap will too. Rtabmap is dependent of tango only for getting the pose and the RGBD images, for anything else it could still evolve on its own. Note that current databases taken with the tango will always be compatible with desktop/ros versions and even possible future ports on ARCore or ArKit.

Cheers,
Mathieu

thibaultmol, 2017-12-17:
Awesome! Thanks

closed 2017-12-17 by thibaultmol