introlab/rtabmap #251: rtabmap gui not showing up

ruchikoff, 2017-12-18:
I followed the rtabmap_ros hand-held mapping instrctions for Kinect XBOX 360 and ran the following:
 roslaunch openni_launch openni.launch depth_registration:=true
and then:
roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start"
But still the GUI did not show up. I tried using the rviz too by running the following:
roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" rviz:=true rtabmapviz:=false , it just creates the first thing it sees on and then does not add up anything further. Please help me with this.

matlabbe, 2017-12-18:
Hi, are there errors in the terminal when launching with rtabmapviz?

ruchikoff, 2017-12-19:
Yes, 
I am getting an error message that says this:
""
[ERROR] [1513677513.240606866]: Skipping XML Document "/opt/ros/indigo/share/gmapping/nodelet_plugins.xml" which had no Root Element.  This likely means the XML is malformed or missing.

""

matlabbe, 2017-12-19:
Can you show rqt_graph after starting the launch?

ruchikoff, 2017-12-19:
This is the rqt_graph that is showing up:

[image]

Also, I tried to just start the GUI by writing rtabmap in the terminal. It showed the following error message:
Segmentation fault (core dumped)

matlabbe, 2017-12-19:
Is it a rtabmap built from source? It seems to be a wrong version of a third party library opened on start-up, causing a crash on start. What kind of machine do you have? 

Not sure what means n_rtabmap_rtabmap and n_rtabmap_rtabmapviz, is rtabmap node also crashed?

ruchikoff, 2017-12-25:
I tried formatting my laptop and installing ROS again. Now it works.

closed 2017-12-25 by ruchikoff