introlab/rtabmap #252: Android compatibility issue

renevanderlende, 2017-12-19:
Hi,

App is not compatible with Samsung Galaxy Edge 7 (nougat)

and -->

Google News:

The Tango project will be deprecated on March 1st, 2018.
Google is continuing AR development with ARCore, a new platform designed for building augmented reality apps for a broad range of devices without the requirement for specialized hardware.

https://developers.google.com/tango/

matlabbe, 2017-12-19:
Google tango is required to run the app. Why it is not working with ARCore right know is because a depth sensor is required, so ARCore won't be supported in the short-term.

TheBricktop, 2019-10-27:
Ive seen other project that reuses goolge tango codebase with ARcore for SLAM and SFM its on github [here]( https://github.com/lvonasek/tango/wiki/3D-Scanner-for-ARcore
) <=link to github.
He even released compiled plugin on Unity Asset store.
So its not possible to create a map using just a point cloud stream from arcore?

matlabbe, 2019-11-03:
Yes, but the point cloud would be sparse, not as dense as on Tango (in particular on textureless surfaces). Some approaches can generate dense depth images in real-time from a single camera (Structure-from-Motion), but the applications are often very limited to environments with a lot of textures and for limited camera movements (in general pure rotations should be avoided). You may take a look at Photogrammetry apps, maybe there are some having already Unity compatibility. Without a depth camera, it is very difficult to have a nice real-time 3D reconstruction feedback like on tango that is very easy to use for anyone.

TheBricktop, 2019-11-04:
I wonder if adding IR grid projector to phone would do any good for feature aquisition in arcore.