introlab/rtabmap #255: FlyCapture2 and Triclops detection fails on Ubuntu 14.04

zlacelle, 2018-01-03:
I think the cmake_modules/FlyCapture2.cmake file doesn't work with Flycapture2 and Triclops on Ubuntu 14.04 (versions: Flycapture2 2.9.3.43, triclops3 3.4.0.8).  The cmake file looks in FlyCapture2_ROOT_DIR/include for the FlyCapture2.h file, but it should also look in FlyCapture2_ROOT_DIR/include/flycapture.  Likewise, for Triclops, it should look in both Triclops_ROOT_DIR/include and Triclops_ROOT_DIR/include/triclops.

Here's a patch to make this work, at least in the "standard" installation of supported versions of flycapture2 and triclops3 on Ubuntu 14.04:

```
diff --git a/cmake_modules/FindFlyCapture2.cmake b/cmake_modules/FindFlyCapture2.cmake
index d96bc15..25c00ca 100644
--- a/cmake_modules/FindFlyCapture2.cmake
+++ b/cmake_modules/FindFlyCapture2.cmake
@@ -19,13 +19,13 @@ set(Triclops_LIBDIR $ENV{Triclops_ROOT_DIR}/lib)
 endif()
 
 #FlyCapture2 SDK
-find_path(FlyCapture2_INCLUDE_DIR NAMES FlyCapture2.h PATHS $ENV{FlyCapture2_ROOT_DIR}/include)
-find_library(FlyCapture2_LIBRARY NAMES FlyCapture2_v100 FlyCapture2 flycapture2 NO_DEFAULT_PATH PATHS ${FlyCapture2_LIBDIR})
+find_path(FlyCapture2_INCLUDE_DIR NAMES FlyCapture2.h PATHS $ENV{FlyCapture2_ROOT_DIR}/include $ENV{FlyCapture2_ROOT_DIR}/include/flycapture)
+find_library(FlyCapture2_LIBRARY NAMES FlyCapture2_v100 FlyCapture2 flycapture2 flycapture NO_DEFAULT_PATH PATHS ${FlyCapture2_LIBDIR})
 
 # Triclops SDK
-find_path(Triclops_INCLUDE_DIR NAMES triclops.h PATHS $ENV{Triclops_ROOT_DIR}/include)
-find_library(Triclops_LIBRARY NAMES triclops triclops_v100 NO_DEFAULT_PATH PATHS ${Triclops_LIBDIR})
-find_library(FlyCaptureBridge_LIBRARY NAMES flycapture2bridge flycapture2bridge_v100 NO_DEFAULT_PATH PATHS ${Triclops_LIBDIR})
+find_path(Triclops_INCLUDE_DIR NAMES triclops.h PATHS $ENV{Triclops_ROOT_DIR}/include $ENV{Triclops_ROOT_DIR}/include/triclops)
+find_library(Triclops_LIBRARY NAMES triclops triclops_v100 libtriclops.so.3 NO_DEFAULT_PATH PATHS ${Triclops_LIBDIR})
+find_library(FlyCaptureBridge_LIBRARY NAMES flycapture2bridge flycapture2bridge_v100 libflycapture2bridge.so.3 NO_DEFAULT_PATH PATHS ${Triclops_LIBDIR})
 find_library(pnmutils_LIBRARY NAMES pnmutils pnmutils_v100 NO_DEFAULT_PATH PATHS ${Triclops_LIBDIR})
 
 IF (FlyCapture2_INCLUDE_DIR AND Triclops_INCLUDE_DIR AND FlyCapture2_LIBRARY AND Triclops_LIBRARY AND FlyCaptureBridge_LIBRARY AND pnmutils_LIBRARY
```

closed 2018-01-04 by matlabbe

zlacelle, 2018-01-04:
Can you merge this into the indigo-devel branch as well?  This is the one I've tested the patch on.

Thanks!

matlabbe, 2018-01-04:
indigo-devel branch is synchronized with the official version released in ROS Indigo. It is used more like a tag than a actual branch (ROS release system requires to link packages to a branch, not a tag). It is recommended to use default branch (master) when building from source. indigo-devel branch will be updated when new Indigo binaries will be released.

commit 79c38d66 (2018-01-04): Fixed #255