introlab/rtabmap #256: Compiling error: ‘>>’ should be ‘> >’ within a nested template argument list

Sagexs, 2018-01-05:
Hey, 

I use ubuntu 16.04 , GNU 5.4.0 

if I try to compile the library I get the following error: 

```
DBDriver.cpp:1064:86: error: ‘>>’ should be ‘> >’ within a nested template argument list
     std::vector<std::vector<Eigen::Vector2f, Eigen::aligned_allocator<Eigen::Vector2f>> > * texCoords,
                                                                                      ^
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:163: die Regel für Ziel „corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o“ scheiterte
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o] Fehler 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
/home/***/library/rtabmap/corelib/src/Memory.cpp:1704:85: error: ‘>>’ should be ‘> >’ within a nested template argument list
    std::vector<std::vector<Eigen::Vector2f, Eigen::aligned_allocator<Eigen::Vector2f>> > * texCoords,
```

I added the following line to the cmakelists .txt to solve the problem:

` SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")`

matlabbe, 2018-01-05:
This is indeed a bug, it should build without c++11. We need to replace all ">>" by "> >".

eric-schleicher, 2018-03-06:
I'm having this same issue with tag/0.14.0

matlabbe, 2018-03-06:
I already fixed some in devel branch (example [this](https://github.com/introlab/rtabmap/blob/master/corelib/src/DBDriver.cpp#L1073) vs [this](https://github.com/introlab/rtabmap/blob/devel/corelib/src/DBDriver.cpp#L1084)). We will see if this problem is still there when I will merge devel to master.