introlab/rtabmap #257: the curious about utility

Gatsby23, 2018-01-09:
First, I would apologize for asking this stupid question. I'm a noob to the SLAM Program. When I read the code of RTABMAP, I don't know how to use the Utilite Library, there's no document on the website. Did you write it by yourself? Could you show me how to understand and use this library?

matlabbe, 2018-01-09:
Yes, UtiLite is a cross-platform library I did back in the days (before c++11 kicked out). The original library [was on google code](https://code.google.com/archive/p/utilite/), which doesn't exist anymore. It has been backup on github here: https://github.com/matlabbe/utilite. If you clone the repository, you could browse the API documentation from the index.html in this subfolder: https://github.com/matlabbe/utilite/tree/master/doc/html

[image]

For convenience, UtiLite library has been copied directly in rtabmap repository as a sub module, which would be slightly more up to date than the version on https://github.com/matlabbe/utilite (I should do a sync someday with the rtabmap version, and also make the API documentation online back again).

If you browse the documentation, there are examples for the main classes like UEventsManager, UThread, ULogger and UPlot.

cheers,
Mathieu

Gatsby23, 2018-01-10:
Dear Editor:
                   Thank you for your prompt reply. That's so sweet!
                                                                                                                                                                                                                                                                                                                                                                                                                                                                    Yours
                                                                                                                                                                                                                                                                                                                                                                                                                                                                    Wu Qi

pet_wolf23@163.com
 
From: matlabbe
Date: 2018-01-10 00:10
To: introlab/rtabmap
CC: Gatsby23; Author
Subject: Re: [introlab/rtabmap] the curious about utility (#257)
Yes, UtiLite is a cross-platform library I did back in the days (before c++11 kicked out). The original library was on google code, which doesn't exist anymore. It has been backup on github here: https://github.com/matlabbe/utilite. If you clone the repository, you could browse the API documentation from the index.html in this subfolder: https://github.com/matlabbe/utilite/tree/master/doc/html
For convenience, UtiLite library has been copied directly in rtabmap repository as a sub module, which would be slightly more up to date than the version on https://github.com/matlabbe/utilite (I should do a sync someday with the rtabmap version, and also make the API documentation online back again).
If you browse the documentation, there are examples for the main classes like UEventsManager, UThread, ULogger and UPlot.
cheers,
Mathieu