introlab/rtabmap #258: How to output VO and loop closure at the same time

forthealllight, 2018-01-21:
Hi， In my experiment ,I want to use my own graph optimization method。But  I want to use your VO and Loop closure in KITTI database 。 So  is there a method in your code ,to ouput the VO and Loop closure in a  ".g2o"  file  when  I give the KITTI database as a input。

thank you~

matlabbe, 2018-01-22:
To get all poses of odometry recorded in the database, you should set "Rtabmap/CreateIntermediateNodes" to true (in GUI, it is in Preferences->RTAB-Map (advanced)). Then, after mapping, we can do File->Export Poses... in g2o format (it will contain odometry poses + loop closure constraints).

While still under construction, there are some infos on this [page](https://github.com/introlab/rtabmap/wiki/Benchmark-SLAM#kitti-dataset) about configuring RTAB-Map for KITTI dataset. However, these parameters are not optimal, I will update this in next months. 

cheers,
Mathieu

forthealllight, 2018-02-06:
thank you ,but when I use KITTI_00 dataset as  the input , to run RTAB-MAP。KITTI_00 has 4540 frames，but loop closure detection is only handled in the first 107 frames。Furthermore when I export the poses .. in g2o format 。There is not 4540 poses,but only 12 poses，I don  not  know why?
I want to use your code ,to gei the full 4540 poses and all loop closure。What should I do?

matlabbe, 2018-04-25:
It depends on the version of RTAB-Map you are using, there was a bug when optical flow is used (Vis/CorType=1) on visual odometry and rtabmap at the same time, causing the warning you are seeing and rejecting all loop closures. In latest version, this has been fixed. 

To make sure to process all frames, you can set the buffer size of rtabmap and odometry to 0 (infinite). Note that you may keep detection rate at 1 Hz, to compute loop closure detection only each 10 frames. Depending on your computer power, setting input rate to 10 or 5 can be good too show results live without lagging. Adjusting Preferences->3D Rendering to show less data (e.g., increasing decimation) can also help for fast rendering.

If you don't care about online visualization, there is now a tool called **rtabmap-kitti_dataset** to process KITTI sequences in an offline manner for convenience.

cheers,
Mathieu