introlab/rtabmap #259: Cannot open librtabmap_gui.so.0.15

Gatsby23, 2018-01-28:
when I run rtabmap in my Ubuntu16.04, it tells me that there are no librtabmap_gui.so.0.15, I found there are only librtabmap_gui.so.0.14 on my computer, How can I fix it?

matlabbe, 2018-01-28:
How it was built? From source or binaries have been also installed?

Gatsby23, 2018-01-29:
Ｍaybe because I installed Rtabmap-0.14 from source before, but I have deleted from usr/local/lib and /usr/local/include ? Are there something else I should do?

Gatsby23, 2018-01-29:
Dear Doctor:
        I'm a master from China. Recently I have read your paper about the RTABMAP. It's really a wonderful and excellent for loop closure detection researcher. But there are question when I read the source code,  why didn't you make the tf-idf strategy for visual word weights update? Or you have made it but I can't get the point, could you please show me your strategy for the weight of viusal word? Thanks!    
                                                                                                                                                                Yours
                                                                                                                                                                   Qi

matlabbe, 2018-01-29:
Uninstall previous version, and rebuild new one. You can use also "ldd" or "which" terminal commands to know which files are used and delete them if they are the old ones.

By "weight", the paper refers to how many times a location has been seen consecutively. This weight is used for WM<->LTM memory management.

The weight you are referring for tf-idf is the weight used to distinguish discriminative visual words in order to estimate better the likelihood of a new location in comparison of the all other (likelihood of a loop closure). In the [original paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/b/bc/TRO2013.pdf) about only loop closure detection, we didn't use tf-idf weighting approach, but a direct comparison between locations to set their likelihood. However, for computation efficiency and that in metric SLAM we can verify if a loop closure is good (e.g., more robust to wrong loop closures), the current default implementation from this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/8/87/LabbeAURO2017.pdf) uses tf-idf approach to evaluate the likelihood.

cheers,
Mathieu