introlab/rtabmap #261: How Can I Understand Robot's Location and How To Move?

mustafamertunali, 2018-01-29:
Hello! I'm beginner at mapping. 

I did my first mapping and I would like to know my robot's location. 
 Can I see the location? 
Can I use that system with Python?
Have a nice day!

matlabbe, 2018-01-29:
In standalone version, we can only "see" in the map view the location of the camera (the small coordinate frame).

I would recommend the rtabmap_ros version if you want to access information outside rtatmap in real-time. With rtabmap_ros, you will be able to know the current pose with TF. As ROS has python interface, you will be able to subscribe to all rtabmap topics in python too. If you want to reproduce what you just did with the standalone version, see [this rtabmap_ros tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping).

cheers,
Mathieu

mustafamertunali, 2018-01-31:
What is the problem now sir? When I try `roslaunch freenect_launch freenect.launch depth_registration:=true`
there is no camera view on the screen. What should I do now?

[image]

matlabbe, 2018-01-31:
Follow next steps:
```
$ roslaunch freenect_launch freenect.launch depth_registration:=true
$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start"
```

mustafamertunali, 2018-02-02:
It is working now but I still can't reach the location.
[image]

matlabbe, 2018-02-02:
Not sure to understand "cannot reach the location", but to see actual pose in map frame, do:
```bash
$ rosrun tf tf_echo map camera_link
```

If you want to use TF in your code, look at these tutorials: http://wiki.ros.org/tf/Tutorials

Cheers,
Mathieu