introlab/rtabmap #263: There are another problem with my RTABMAP, I guess it is caused by the opencv and qt5, and I don't know how to make it? Is there are another way to solve this without reinstall Ubuntu?

Gatsby23, 2018-02-01:
*** Error in `rtabmap': realloc(): invalid pointer: 0x00007ff8bac46000 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7ff8b92f27e5]
/lib/x86_64-linux-gnu/libc.so.6(realloc+0x348)[0x7ff8b92ffa08]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QListData7reallocEi+0x1f)[0x7ff8b9ce79cf]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QListData6appendEi+0x81)[0x7ff8b9ce7aa1]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x1d6d78)[0x7ff8b9db3d78]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_Z21qRegisterResourceDataiPKhS0_S0_+0x2e6)[0x7ff8b9dafb16]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x7bcc3)[0x7ff8b9c58cc3]
/lib64/ld-linux-x86-64.so.2(+0x106ba)[0x7ff8c10c66ba]
/lib64/ld-linux-x86-64.so.2(+0x107cb)[0x7ff8c10c67cb]
/lib64/ld-linux-x86-64.so.2(+0xc6a)[0x7ff8c10b6c6a]

matlabbe, 2018-02-01:
Which rtabmap version? Binaries/source? Ubuntu 16.04? Is this crash happen on start or while mapping?

Gatsby23, 2018-02-02:
The source file, it's crash when it happen

发自网易邮箱大师

matlabbe, 2018-02-02:
Is it crashing before the window shows up?

QinZiwen, 2018-03-13:
@Gatsby23 Is this problem solved?