introlab/rtabmap #264: Using low cost stereo cameras

patrickelectric, 2018-02-02:
I am using a stereo camera like [this one](https://pt.aliexpress.com/item/Megapixel-MJPEG-30fps-2560-720-Sony-IMX179-UVC-Plug-Play-Driverless-73mm-x-23mm-small-video/32699038862.html), the output of both cameras is a mjpg in /dev/video1 and /dev/video2.

Is it possible to use rtabmap with this kind of setup (without using ROS) ?

matlabbe, 2018-02-02:
Without ros, rtabmap supports stereo USB camera, as.long as the camera can stream side by side images. To read from two input sources, a new driver would have to be created, which could be similar to [CameraStereoVideo](https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/CameraStereo.h#L201) class. However, stereo images should be exactly symchronized, so it is preferred that the camera is able to stream already synchronized images.

Cheers,
Mathieu

patrickelectric, 2018-02-02:
Ty @matlabbe, I'll be very glad to write a new driver for rtabmap !
I believe that scenario is pretty common, and this new drive can be util.
My best regards

matlabbe, 2018-02-02:
The CameraStereoVideo class with two videos as input can be very close to what you want to achieve. For example, this [code](https://github.com/introlab/rtabmap/blob/d7871fec2b305e10d4261f3f8be57e995e406c69/corelib/src/CameraStereo.cpp#L1345-L1347):
```cpp
capture_.open(path_.c_str());
capture2_.open(path2_.c_str());
```
could be replaced by:
```cpp
capture_.open(usbDevice1);
capture2_.open(usbDevice2);
```
**However**, why the original code doesn't support 2 usb devices but just 2 videos is that there is no certainty that cv::VideoCapture called on two devices will give the frames at the same time. Timing is very important for stereo. Look if your stereo camera has already a driver that can fetch both images at the same time, then if yes, you may create another CameraStereo[MyDriverName] class based on this driver (like CameraStereoZed for ZED camera or CameraStereoFlyCapture2 for Point grey cameras).

If the camera is a standard UVC camera, it is already compatible with rtabmap (through OpenCV cv::VideoCapture).

cheers,
Mathieu

paulsammut, 2018-02-12:
As a suggestion, I am using the tara's from E-Con systems and for 250 bucks they are excellent. Baseline is a little small at 6cm, but for indoor use, range from 0.3cm to 3m the data they return is great. Better than ZED at that range. 

They have a decent ROS driver and work well with rtabmap. Very very very worth the little amount of money they cost when compared to any other decent stereo rig.

jm677zz, 2018-03-07:
[Tara](https://www.e-consystems.com/3D-USB-stereo-camera.asp) has been very widely in outdoor applications where sunlight is there.  E-con Systems provides a C++ SDK for Windows and Linux and includes some standard analysis examples using OpenCV such as height estimation, face detection, and point cloud generation.