introlab/rtabmap #267: libfreenect2 not found

osakseid, 2018-02-28:
I'm trying to use Kinect v2 with rtabmap.

I'm using Freenect2 (https://github.com/OpenKinect/libfreenect2/blob/master/README.md)
When installing a stand alone version of rtabmap (https://github.com/introlab/rtabmap/wiki/Installation#ubuntu) and running "make .." i'm getting "libfreenect2 not found".

I specify "cmake -Dfreenect2_DIR=$HOME/freenect2/lib/cmake/freenect2" but Cmake doesn't seems to find Freenect2.

osakseid, 2018-02-28:
The kinect2 works fine with Protonect.

matlabbe, 2018-03-01:
Is there a freenect2Config.cmake in $HOME/freenect2/lib/cmake/freenect2?

Can you show the output of cmake from rtabmap?

rui-cao, 2018-03-16:
I have the exact same problem.
Here is my cmake compilation report (freenect2 is detected):
[image]

The error I get when I open the rtabmap is like this:
[image]

Thanks

matlabbe, 2018-03-16:
The error refers to CameraFreenect, not CameraFreenect2, make sure you selected Freenect2 driver (not Freenect) under "Kinect v2" sensor.

cheers

rui-cao, 2018-03-17:
I am aware of that. Here is a screen-shoot of me trying to enable the freenect2 option, however it is not available. Even though it is detected in cmake.

[image]

Thanks

matlabbe, 2018-03-17:
Is it the rtabmap you built that is launched? (`$ which rtabmap`)

rui-cao, 2018-03-18:
Thank you! The problem is solved.
By running $ which rtabmap, it shows the rtabmap from the ros package directory instead of my compiled-from-source directory. I uninstalled the ros package, and now it is working with the freenect2.

Luke1962, 2021-12-27:
Also in my case Freenec2 connection was disabled, due to library not found by cmake.
I've solved by adding to .bashrc file
`export CMAKE_PREFIX_PATH=/home/<my user name>/freenect2/lib/cmake/freenect2:$CMAKE_PREFIX_PATH`