introlab/rtabmap #268: curious error preventing recompile to latest rtabmap

eric-schleicher, 2018-03-05:
[image]

I've gotten the following error whilst attempting to update rtabmap (proper) from current master.  
the error is peculiar, given the while ROS is installed on this system I'm not attempting to build the rtabmap_ros library.   so the rule entry seems confusing to me.

Here is a [full gist of the log output](https://gist.github.com/eric-schleicher/625a261706fb68dbf6f843bcecc6e56c)

I had previously had the realsense sdk installed from source and build rtabmap with support; however for an older.  I've since remove it and installed the latest version (not from inside a ros workspace, but from a user home directory)  `e.g. ~/librealsense` and completed the `make && sudo make install` steps which installs itself in the `usr/local/bin/'  i can run the tests and turn the camera on and off as expected running the realsense2 SDK.

in reviewing the output of cmake, it appears that it's "finding" the realsense library in  `/opt/ros/kinectic/include`   however it's not installed (that i can tell).

is it looking for the library based on an environment variable? or actually finding some files?

matlabbe, 2018-03-05:
It may be still in CMakeCache.txt in the build directory of rtabmap. Try a clean build:
```bash
$ rm -r rtabmap/build/*
$ cd rtabmap/build
$ cmake .. 
```
If realsense is still found but it is not installed, it seems cmake finds a realsense config somewhere. To explicitly disable realsense:
```bash
$ cmake -DWITH_REALSENSE=OFF .. 
```
If new realsense sdk is installed, I am not sure if the current [FindRealSense.cmake](https://github.com/introlab/rtabmap/blob/master/cmake_modules/FindRealSense.cmake) will find it, causing problems... so explicitly disable realsense as above would be also required.

eric-schleicher, 2018-03-06:
hi @matlabbe I did as you mentioned here, but it took a few additional attempts to get a succeful build of rtabmap...  I ended up having to:

- remove (`make uninstall`)the realsense(1 and 2) libraries (IIRC if left in rtabmap segfaults on start)
- manually add the stdC++11 flag to get rtabmap (0.14.0) same as in this [issue](https://github.com/introlab/rtabmap/issues/256)
- `make uninstall` PCL 1.8.

after this ran into some issues starting it (referencing missing libraries), and then manually cleaned out the rtabmap files in `usr/local/rtab*` as there were a bunch of old version.   after this i was able to make install it and verified it.

HOWEVER now the issue is in rtabmap_ros where in order to get other packages to compile i need to have pcl 1.8 installed (or set their pcl_include_dirs; which i don't know how to do).

now with catkin_make, compilaiton starts generating errors when it gets into rtabmap_ros . 
 here is a -j1 [output log file](https://gist.github.com/eric-schleicher/f9206006b9d539ef8439c4bbaa512c2e#file-2018-03-06-catkin_make-log-L570) highlighting where we start generating errors.

Are these issues because the base rtabmap is compiled with pcl 1.7? and rtabmap_ros dependency (perception_pcl) is set for pcl 1.8?

I'm just trying to get a clear path where each package can use what it needs.  I *have* to build pcl 1.8 in the workspace for OpenChisel package.   rtabmap is picking that up.

matlabbe, 2018-03-06:
If you don't want to uninstall pcl 1.7 (because many stuff may depend on it), you would have to force rtabmap to use 1.8 [here](https://github.com/introlab/rtabmap/blob/71d9816f4bb6e47bb2a90fb3e82cf911f30a2b12/CMakeLists.txt#L174-L176) (as rtabmap_ros depends on pcl_ros which would be built manually with 1.8 to support open_chisel if I understand).

If rtabmap is built with g2o, gtsam, zed or realsense support, it will automatically add c++11 to compilation, see [here](https://github.com/introlab/rtabmap/blob/71d9816f4bb6e47bb2a90fb3e82cf911f30a2b12/CMakeLists.txt#L418-L432).

For the error log, make sure to use same version/branch for both rtabmap and rtabmap_ros. It is not a problem with 1.7 or 1.8, rtabmap/rtabmap_ros should build fine with each other.

eric-schleicher, 2018-03-07:
I was ultimately able to get rtabmap to compile with 1.8 it's possible that CPUTSDF PCL dependency was interferening?  would that even make sense?

closed 2018-03-08 by eric-schleicher

matlabbe, 2018-03-08:
CPUTSDF should have been forced to be built also with PCL 1.8 if so.

eric-schleicher, 2018-03-08:
it had been built previously, independently.