introlab/rtabmap #269: cannot find -lopencv_dep_cudart

QinZiwen, 2018-03-12:
when cmake it has a warning, but when make , the terminal show "/usr/bin/ld: cannot find -lopencv_dep_cudart"
```
$ cmake ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found suitable exact version "8.0") 
-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.4
-- Found Eigen: /usr/include/eigen3  
-- Eigen found (include: /usr/include/eigen3, version: 3.2.0)
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.20
-- Found USB_10: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- Found libusb-1.0: /usr/include  
-- Checking for module 'flann'
--   Found flann, version 1.8.4
-- Found FLANN: /usr/lib/x86_64-linux-gnu/libflann_cpp.so  
-- FLANN found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp.so;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp.so)
-- FLANN found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp.so;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp.so)
-- Found Qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- Found PCL_COMMON: /usr/local/lib/libpcl_common.so  
-- Found PCL_OCTREE: /usr/local/lib/libpcl_octree.so  
-- Found PCL_IO: /usr/local/lib/libpcl_io.so  
-- Found PCL_KDTREE: /usr/local/lib/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/local/lib/libpcl_search.so  
-- Found PCL_SURFACE: /usr/local/lib/libpcl_surface.so  
-- Found PCL_FILTERS: /usr/local/lib/libpcl_filters.so  
-- Found PCL_2D: /usr/local/include/pcl-1.8  
-- Found PCL_FEATURES: /usr/local/lib/libpcl_features.so  
-- Found PCL_REGISTRATION: /usr/local/lib/libpcl_registration.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/local/lib/libpcl_sample_consensus.so  
-- Found PCL_GEOMETRY: /usr/local/include/pcl-1.8  
-- Found PCL_ML: /usr/local/lib/libpcl_ml.so  
-- Found PCL_SEGMENTATION: /usr/local/lib/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/local/lib/libpcl_visualization.so  
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
-- Found OpenCV: /usr/local/include/opencv;/usr/local/include
-- Found PCL: /usr/local/include/pcl-1.8;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/local/include/vtk-6.3
-- Found ZLIB: /usr/include
[... 195 lines cut ...]
[ 33%] Building C object corelib/src/CMakeFiles/rtabmap_core.dir/sqlite3/sqlite3.c.o
[ 33%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/discrete_depth_distortion_model_helpers.cpp.o
[ 34%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/discrete_depth_distortion_model.cpp.o
[ 34%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/frame_projector.cpp.o
[ 34%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/slam_calibrator.cpp.o
[ 35%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/opencv/Orb.cpp.o
[ 35%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/opencv/solvepnp.cpp.o
[ 36%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/toro3d/posegraph3.cpp.o
[ 36%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/toro3d/treeoptimizer3_iteration.cpp.o
[ 36%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/toro3d/treeoptimizer3.cpp.o
[ 37%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/toro3d/posegraph2.cpp.o
[ 37%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/toro3d/treeoptimizer2.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/vertigo/g2o/edge_se2Switchable.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/vertigo/g2o/edge_se3Switchable.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/vertigo/g2o/edge_switchPrior.cpp.o
[ 39%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/vertigo/g2o/types_g2o_robust.cpp.o
[ 39%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/vertigo/g2o/vertex_switchLinear.cpp.o
[ 40%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OctoMap.cpp.o
[ 40%] Linking CXX shared library ../../../bin/librtabmap_core.so
/usr/bin/ld: cannot find -lopencv_dep_cudart
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/librtabmap_core.so.0.15.0] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
```

matlabbe, 2018-03-12:
It seems you have built OpenCV with CUDA dependency. Make sure libopencv_dep_cudart.so is in a path of LD_LIBRARY_PATH.

QinZiwen, 2018-03-13:
After opencv compile and install, there is no libopencv_dep_cudart.so in /usr/local/lib

QinZiwen, 2018-03-13:
Fix this problem https://github.com/opencv/opencv/issues/6542
```
cmake -D CUDA_USE_STATIC_CUDA_RUNTIME=OFF ..
```

No error occurred during compilation， But there was an invalid pointer error during the test.
```
rtabmap/bin$ ./rtabmap
*** Error in `./rtabmap': realloc(): invalid pointer: 0x00007f1e8d3aa840 ***
Aborted (core dumped)
rtabmap/bin$ ./rtabmap-databaseViewer 
*** Error in `./rtabmap-databaseViewer': realloc(): invalid pointer: 0x00007f3fc0660840 ***
Aborted (core dumped)
```