introlab/rtabmap #270: segfault using rtabmap on ubuntu 17.10

kaushik-rohit, 2018-03-19:
I am trying to use rtabmap on Ubuntu 17.10 for my project and have compiled it from source. But the problem is it segfaults every time. I am using freenect driver for Kinect.

```
rohit@rohit-Inspiron-5577:~/projects/rtabmap/build$ rtabmap
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ WARN] (2018-03-19 21:35:36.761) Parameters.cpp:836::readINI() Section "Core" in /home/rohit/.rtabmap/rtabmap.ini doesn't exist... Ignore this warning if the ini file does not exist yet. The ini file will be automatically created when rtabmap will close.
[ WARN] (2018-03-19 21:35:59.365) CameraModel.cpp:314::load() Could not load calibration file "/home/rohit/Documents/RTAB-Map/camera_info/B00363710742144B_depth.yaml".
[ WARN] (2018-03-19 21:35:59.365) CameraModel.cpp:314::load() Could not load calibration file "/home/rohit/Documents/RTAB-Map/camera_info/B00363710742144B_rgb.yaml".
[ WARN] (2018-03-19 21:35:59.365) CameraRGBD.cpp:1228::init() Missing calibration files for camera "B00363710742144B" in "/home/rohit/Documents/RTAB-Map/camera_info" folder, default calibration used.
Segmentation fault (core dumped)

```

matlabbe, 2018-03-19:
Is it when you click on start detection the issue happens? If so, you can change logger level in Preferences->Logger to Debug to show more info before it crashes. You can also launch with gdb to debug:
```
$ gdb rtabmap
[type "run"]
[rtabmap opens, click on "Start"]
[seg fault]
[type "bt" in gdb and copy/paste the log here]
```

cheers,
Mathieu

kaushik-rohit, 2018-03-20:
```
rohit@rohit-Inspiron-5577:~/projects/mrpt/build$ rtabmap
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ WARN] (2018-03-21 02:41:52.386) Parameters.cpp:836::readINI() Section "Core" in /home/rohit/.rtabmap/rtabmap.ini doesn't exist... Ignore this warning if the ini file does not exist yet. The ini file will be automatically created when rtabmap will close.
[DEBUG] (2018-03-21 02:42:06.797) PreferencesDialog.cpp:1363::closeDialog() 
[DEBUG] (2018-03-21 02:42:15.829) MainWindow.cpp:4380::newDatabase() 
[DEBUG] (2018-03-21 02:42:15.830) CloudViewer.cpp:1047::removeOctomap() 
[ INFO] (2018-03-21 02:42:17.293) MainWindow.cpp:4397::newDatabase() Deleted temporary database "/home/rohit/Documents/RTAB-Map/rtabmap.tmp.db".
[DEBUG] (2018-03-21 02:42:17.294) RtabmapThread.cpp:400::handleEvent() CMD_INIT
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3966::applyPrefSettings() 
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=BRIEF/Bytes Value=32
[DEBUG] (2018-03-21 02:42:17.294) RtabmapThread.cpp:71::pushNewState() to 0
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=BRISK/Octaves Value=3
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=BRISK/PatternScale Value=1
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=BRISK/Thresh Value=30
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=Bayes/FullPredictionUpdate Value=false
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=Bayes/PredictionLC Value=0.1 0.36 0.30 0.16 0.062 0.0151 0.00255 0.000324 2.5e-05 1.3e-06 4.8e-08 1.2e-09 1.9e-11 2.2e-13 1.7e-15 8.5e-18 2.9e-20 6.9e-23
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=Bayes/VirtualPlacePriorThr Value=0.9
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=DbSqlite3/CacheSize Value=10000
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=DbSqlite3/InMemory Value=false
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=DbSqlite3/JournalMode Value=3
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=DbSqlite3/Synchronous Value=0
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=DbSqlite3/TempStore Value=2
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FAST/Gpu Value=false
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FAST/GpuKeypointsRatio Value=0.05
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FAST/GridCols Value=4
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FAST/GridRows Value=4
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FAST/MaxThreshold Value=200
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FAST/MinThreshold Value=7
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FAST/NonmaxSuppression Value=true
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FAST/Threshold Value=20
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FREAK/NOctaves Value=4
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FREAK/OrientationNormalized Value=true
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FREAK/PatternScale Value=22
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=FREAK/ScaleNormalized Value=true
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=GFTT/BlockSize Value=3
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=GFTT/K Value=0.04
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=GFTT/MinDistance Value=3
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=GFTT/QualityLevel Value=0.001
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=GFTT/UseHarrisDetector Value=false
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=GTSAM/Optimizer Value=1
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=Grid/3D Value=true
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=Grid/CellSize Value=0.05
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=Grid/ClusterRadius Value=0.1
[DEBUG] (2018-03-21 02:42:17.294) MainWindow.cpp:3972::applyPrefSettings() Parameter changed: Key=Grid/DepthDecimation Value=4
[... cut ...]
[ INFO] (2018-03-21 02:42:26.317) Rtabmap.cpp:2352::process() timeMapOptimization=0.000008s
[ INFO] (2018-03-21 02:42:26.317) Rtabmap.cpp:2379::process() sending stats...
[ INFO] (2018-03-21 02:42:26.317) Rtabmap.cpp:2389::process() send all stats...
[ INFO] (2018-03-21 02:42:26.317) Rtabmap.cpp:2422::process() Set map correction = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[ INFO] (2018-03-21 02:42:26.317) Rtabmap.cpp:2482::process() Time creating stats = 0.000089...
Segmentation fault (core dumped)
```

kaushik-rohit, 2018-03-20:
The gdb log as well : 
```
rohit@rohit-Inspiron-5577:~/projects/mrpt/build$ gdb rtabmap
GNU gdb (Ubuntu 8.0.1-0ubuntu1) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rtabmap...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/local/bin/rtabmap 
warning: the debug information found in "/lib64/ld-2.26.so" does not match "/lib64/ld-linux-x86-64.so.2" (CRC mismatch).

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffbccef700 (LWP 9046)]
[New Thread 0x7fffb92c1700 (LWP 9047)]
[New Thread 0x7fffaee1d700 (LWP 9048)]
[New Thread 0x7fffae61c700 (LWP 9049)]
[New Thread 0x7fffad3cc700 (LWP 9050)]
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[New Thread 0x7fff9f9c7700 (LWP 9052)]
[ WARN] (2018-03-21 02:46:36.894) Parameters.cpp:836::readINI() Section "Core" in /home/rohit/.rtabmap/rtabmap.ini doesn't exist... Ignore this warning if the ini file does not exist yet. The ini file will be automatically created when rtabmap will close.
[New Thread 0x7fff9ed1e700 (LWP 9053)]
[New Thread 0x7fff9e51d700 (LWP 9054)]
[New Thread 0x7fff9dd1c700 (LWP 9055)]
[New Thread 0x7fff9d51b700 (LWP 9056)]
[New Thread 0x7fff9cd1a700 (LWP 9057)]
[New Thread 0x7fff9c519700 (LWP 9058)]
[New Thread 0x7fff9bd18700 (LWP 9059)]
[New Thread 0x7fff9b517700 (LWP 9060)]
[ WARN] (2018-03-21 02:46:51.344) CameraModel.cpp:314::load() Could not load calibration file "/home/rohit/Documents/RTAB-Map/camera_info/B00363710742144B_depth.yaml".
[ WARN] (2018-03-21 02:46:51.344) CameraModel.cpp:314::load() Could not load calibration file "/home/rohit/Documents/RTAB-Map/camera_info/B00363710742144B_rgb.yaml".
[ WARN] (2018-03-21 02:46:51.344) CameraRGBD.cpp:1228::init() Missing calibration files for camera "B00363710742144B" in "/home/rohit/Documents/RTAB-Map/camera_info" folder, default calibration used.
[New Thread 0x7fff9a827700 (LWP 9064)]
[New Thread 0x7fff9a026700 (LWP 9065)]
[New Thread 0x7fff99825700 (LWP 9066)]
[New Thread 0x7fff99024700 (LWP 9067)]
[New Thread 0x7fff98823700 (LWP 9068)]
[New Thread 0x7fff7bfff700 (LWP 9069)]
[New Thread 0x7fff7b7fe700 (LWP 9070)]
[New Thread 0x7fff7affd700 (LWP 9071)]
[New Thread 0x7fff7a7fc700 (LWP 9072)]
[New Thread 0x7fff79ffb700 (LWP 9073)]

Thread 15 "rtabmap" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff9b517700 (LWP 9060)]
0x00007ffff6fd50d0 in rtabmap::RtabmapThread::getData(rtabmap::OdometryEvent&) () from /usr/local/lib/librtabmap_core.so.0.16
(gdb) bt
#0  0x00007ffff6fd50d0 in rtabmap::RtabmapThread::getData(rtabmap::OdometryEvent&) () at /usr/local/lib/librtabmap_core.so.0.16
#1  0x00007ffff6fd8f4b in rtabmap::RtabmapThread::process() () at /usr/local/lib/librtabmap_core.so.0.16
#2  0x00007ffff6fda210 in rtabmap::RtabmapThread::mainLoop() () at /usr/local/lib/librtabmap_core.so.0.16
#3  0x00007ffff6c7b922 in UThreadC<void>::ThreadMainHandler(UThreadC<void>::Instance*) () at /usr/local/lib/librtabmap_utilite.so.0.16
#4  0x00007ffff40217fc in start_thread (arg=0x7fff9b517700) at pthread_create.c:465
#5  0x00007ffff4551b5f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) bt full
#0  0x00007ffff6fd50d0 in rtabmap::RtabmapThread::getData(rtabmap::OdometryEvent&) () at /usr/local/lib/librtabmap_core.so.0.16
#1  0x00007ffff6fd8f4b in rtabmap::RtabmapThread::process() () at /usr/local/lib/librtabmap_core.so.0.16
#2  0x00007ffff6fda210 in rtabmap::RtabmapThread::mainLoop() () at /usr/local/lib/librtabmap_core.so.0.16
#3  0x00007ffff6c7b922 in UThreadC<void>::ThreadMainHandler(UThreadC<void>::Instance*) () at /usr/local/lib/librtabmap_utilite.so.0.16
#4  0x00007ffff40217fc in start_thread (arg=0x7fff9b517700) at pthread_create.c:465
        pd = 0x7fff9b517700
        now = <optimized out>
        unwind_buf = 
              {cancel_jmp_buf = {{jmp_buf = {140735799195392, -2008717858781820353, 140737488345950, 140737488345951, 140735799195392, 140735799196096, 2008798272830016063, 2008730521153276479}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#5  0x00007ffff4551b5f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

matlabbe, 2018-03-21:
Thx for the report, I will try to setup a 17.10 machine soon to test this.

kaushik-rohit, 2018-03-21:
@matlabbe Thanks. I am able to run it in docker for now. Can you suggest if there is some tutorial for sending rtabmap data (3d map) over wifi (possibly in real time)? I plan to use it on raspberry pi and send the generated map over wifi to my laptop.

matlabbe, 2018-03-21:
To get data out of RTAB-Map in real-time, I suggest to use its ROS package ([rtabmap_ros](http://wiki.ros.org/rtabmap_ros)) as there is no built-in TCP/IP publisher on the standalone app.

Here is an example running on a RPI: http://official-rtab-map-forum.67519.x6.nabble.com/RGB-D-SLAM-example-on-ROS-and-Raspberry-Pi-3-td1250.html

matlabbe, 2018-03-22:
Fixed by referred commit above.

closed 2018-03-22 by matlabbe