introlab/rtabmap #272: RTABMAP Make ERROR

tonike24, 2018-03-26:
Hey guys, i tried to install RTABMAP and I have the next problem when I execute "Make" in /rtabmap/build.

Scanning dependencies of target euroc_dataset
[ 86%] Building CXX object tools/EurocDataset/CMakeFiles/euroc_dataset.dir/main.cpp.o
/home/javier/rtabmap/tools/EurocDataset/main.cpp: In function ‘int main(int, char**)’:
/home/javier/rtabmap/tools/EurocDataset/main.cpp:381:16: error: ‘stof’ is not a member of ‘std’
       gyr[j] = std::stof(s);
                ^
/home/javier/rtabmap/tools/EurocDataset/main.cpp:387:16: error: ‘stof’ is not a member of ‘std’
       acc[j] = std::stof(s);
                ^
/home/javier/rtabmap/tools/EurocDataset/main.cpp:390:21: error: ‘stoi’ is not a member of ‘std’
      t_imu = double(std::stoi(seconds)) + double(std::stoi(nanoseconds))*1e-9;
                     ^
/home/javier/rtabmap/tools/EurocDataset/main.cpp:390:50: error: ‘stoi’ is not a member of ‘std’
      t_imu = double(std::stoi(seconds)) + double(std::stoi(nanoseconds))*1e-9;
                                                  ^
tools/EurocDataset/CMakeFiles/euroc_dataset.dir/build.make:62: fallo en las instrucciones para el objetivo 'tools/EurocDataset/CMakeFiles/euroc_dataset.dir/main.cpp.o'
make[2]: *** [tools/EurocDataset/CMakeFiles/euroc_dataset.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:984: fallo en las instrucciones para el objetivo 'tools/EurocDataset/CMakeFiles/euroc_dataset.dir/all'
make[1]: *** [tools/EurocDataset/CMakeFiles/euroc_dataset.dir/all] Error 2
Makefile:149: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error 2

This is my Cmake : 

javier@javier-GP62MVR-6RF:~/rtabmap/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found eigen: /usr/include/eigen3  
m-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
a-- Found openni: /usr/lib/libOpenNI.so  
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

k-- Found libusb-1.0: /usr/include  
-- Checking for module 'flann'
--   Found flann, version 1.8.4
e-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
-- Found PCL_COMMON: /usr/lib/x86_64-linux-gnu/libpcl_common.so  
-- Found PCL_OCTREE: /usr/lib/x86_64-linux-gnu/libpcl_octree.so  
-- Found PCL_IO: /usr/lib/x86_64-linux-gnu/libpcl_io.so  
-- Found PCL_KDTREE: /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/lib/x86_64-linux-gnu/libpcl_search.so  
-- Found PCL_SURFACE: /usr/lib/x86_64-linux-gnu/libpcl_surface.so  
-- Found PCL_FILTERS: /usr/lib/x86_64-linux-gnu/libpcl_filters.so  
-- Found PCL_FEATURES: /usr/lib/x86_64-linux-gnu/libpcl_features.so  
-- Found PCL_REGISTRATION: /usr/lib/x86_64-linux-gnu/libpcl_registration.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so  
-- Found PCL_GEOMETRY: /usr/include/pcl-1.7  
-- Found PCL_SEGMENTATION: /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/lib/x86_64-linux-gnu/libpcl_visualization.so  
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
-- Found OpenCV: /opt/ros/kinetic/include/opencv-3.3.1;/opt/ros/kinetic/include/opencv-3.3.1/opencv
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-6.2;/usr/include/jsoncpp;/usr/include/hdf5/openmpi;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/x86_64-linux-gnu;/usr/include/python2.7;/usr/include/libxml2;/usr/include/tcl
-- Found ZLIB: /usr/include
-- Found Freenect: /usr/include
CMake Warning at cmake_modules/FindKinectSDK2.cmake:71 (message):
  Directory "" not found.
Call Stack (most recent call first):
  cmake_modules/FindKinectSDK2.cmake:119 (check_dir)
  CMakeLists.txt:271 (FIND_PACKAGE)

-- KinectSDK2_FOUND : FALSE
-- Could NOT find CSPARSE (missing:  CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY) 
-- Found octomap 1.8.1: /opt/ros/kinetic/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.16.3
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = NO (OpenNI2 not found)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With dc1394               = NO (dc1394 not found)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = NO (g2o not found)
--   With GTSAM                = NO (GTSAM not found)
--   With VERTIGO              = NO (GTSAM or g2o required)
--   With cvsba                = NO (cvsba not found)
--   With libpointmatcher      = NO (libpointmatcher not found)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With OCTOMAP              = YES (License: BSD)
--   With CPUTSDF              = NO (CPUTSDF not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With okvis                = NO (okvis not found)
--   With ORB_SLAM2            = NO (ORB_SLAM2 not found, make sure environment variable ORB_SLAM2_ROOT_DIR is set)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/javier/rtabmap/build

Im not sure how to resolve this, hope you can tell me what is wrong.

Thank you.

tonike24, 2018-03-26:
[100%] Built target noEventsExample

THANK YOU VERY MUCH! I just install : sudo apt-get install ros-kinetic-libg2o and now Make finally end without errors.

littleggghost, 2018-11-03:
Thanks @tonike24 , you are right!