introlab/rtabmap #278: TF tree using tangostreamer with rtabmap

fiml, 2018-04-15:
Hi All,
I am quite new to ROS and rtabmap. I have followed the rtabmap tutorial @http://wiki.ros.org/rtabmap_ros/Tutorials/Tango%20ROS%20Streamer and I am able to successfully map and localize on the map using rtabmapviz. I tried using the launch file demo_turtlebot_tango.launch but that doesn't work. The map quality is poor and I cant get localization to work as well. So I use the command line to map and for localization.

I want to use these results to publish navigation goals to a custom robot. The robot has its own odometry but as mentioned in the tutorial, I have turned it off. But then I am having issues with my TF tree. It doesn't look anything like the one mentioned in the tutorial, ie.  /map -> /odom -> /start_of_service -> /device -> /base_footprint -> /base_link -> ...

I used the tf static_transform_publisher to publish the following lines 
`  <node if="$(arg tango_odometry)" pkg="tf" type="static_transform_publisher" name="device_base_link" args="-0.3 0 0 0 1.5707963267948966 0 device base_footprint 100" /> 
  <node if="$(arg tango_odometry)" pkg="tf" type="static_transform_publisher" name="odom_start_of_service_link" args="0 0 0.3 0 0 0 odom start_of_service 100" /> `

But then my TF Tree looks like this:
[image]

So then i added 
`  <node if="$(arg tango_odometry)" pkg="tf" type="static_transform_publisher" name="map_odom_link" args="0 0 0.3 0 0 0 map odom 100" /> 
  <node if="$(arg tango_odometry)" pkg="tf" type="static_transform_publisher" name="odom_start_of_service_link" args="0 0 0.3 0 0 0 odom start_of_service 100" /> 
  <node if="$(arg tango_odometry)" pkg="tf" type="static_transform_publisher" name="device_base_link" args="-0.3 0 0 0 1.5707963267948966 0 device base_footprint 100" /> `

And then i got the following TF tree:
[image]

I am not sure if this is the desired TF tree. Also the average rate for all broadcasters don't match. Could this be a problem ? 
Finally, I started rviz to see the TF and the frames from the tango were fluctuating. I am not sure what am I doing wrong but I can't localize the robot in rviz or publish a goal. I feel if the TF were to be fine, rest would work. I would be very grateful if someone could review my TF.

Best Regards,
PG

matlabbe, 2018-04-16:
Hi,

Based on [demo_turtlebot_tango.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_turtlebot_tango.launch) and when tango odometry is used (`tango_odometry:=true`), the frames are:

 * `/map -> /odom` should be published by rtabmap node
 * `/odom -> /start_of_service` is a static transform
 * `/start_of_service -> /device` should be published by tango ros streamer
 * `/device -> /base_footprint` is a static transform (would be inverse of `/odom -> /start_of_service`)

Note that in this demo, `odom_frame_id` is set to `odom`, not `start_of_service` like the hand-held example, it is why rtabmap was publishing `map -> start_of_service` instead of `/map -> /odom`. See parameter set [here](https://github.com/introlab/rtabmap_ros/blob/63b6acf031058224ad31beb48774b4457cd990af/launch/demo/demo_turtlebot_tango.launch#L66).

cheers,
Mathieu

fiml, 2018-04-17:
Hi Mathieu,
Thank you for the reply.

I havn't had good mapping/localization results using the demo_turtlebot_tango.launch, so I am launching rtabmap via commandline as in the [tango ros treamer tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/Tango%20ROS%20Streamer). In one of the issues posted @ [Estimated Trajectory is noisy](https://github.com/introlab/rtabmap/issues/213), your solution worked for me too. And there `odom_frame_id-> start_of_service`  and `frame_id->device`. So This shouldn't be the problem.

Although your explanation of transforms really helps my understanding. After I implement the changes you mentioned, my TF tree still looks like 

[image]

On rviz, I could see the tango being able to localize itself but looking at the TF tree, I am wondering if the problem is that the rtabmap doesn't really publish this map-> odom tf. Which would mean my localization doesn't work or doesn't work well. Is this the correct?

Also my robot has its odometry, which I shut off (as mentioned in the [tango ros treamer tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/Tango%20ROS%20Streamer)) . But local planner in the navigation stack requires odometry. Is there any message from rtabmap that could instead be remapped to supply odometry to the navigation stack ?

fiml, 2018-04-17:
I think maybe the following info is also relevant to the problem that I am having. I would really appreciate if you could review these and offer your opinion.

So I am using a 5Ghz band for communication. The robot has a jetson board and runs the roscore. Because of space limitations, I am using a separate computer for running rtabmap and navigation stack. The tango connects also on the same network and send data to my computer. The frequencies for the messages received on my computer that are sent from tango over the following topics are -
rgb_topic:=/tango/camera/color_1/image_raw
average rate: 0.526 min: 1.705s max: 2.703s std dev: 0.17750s window: 54

cloud:=/tango/point_cloud
average rate: 4.948 min: 0.044s max: 0.476s std dev: 0.05741s window: 694

camera_info:=/tango/camera/color_1/camera_info
average rate: 14.718 min: 0.000s max: 0.277s std dev: 0.07166s window: 226

scan:=/tango/laser_scan
average rate: 5.048 min: 0.096s max: 0.253s std dev: 0.03889s window: 18

depth_image:=/tango/registered_depth
average rate: 4.991 min: 0.180s max: 0.213s std dev: 0.00795s window: 25

As I am new to the entire project, I am not sure if these frequencies are even enough for rtabmap.

Best Regards,
Preity

fiml, 2018-04-19:
Hi Mathieu,
Thank you so much for the explaination.
I am still not able to get my robot to move. I don't have a turtlebot but a simple differential drive robot. No sensors on it except wheel encoders. It publishes on the topic /odom. The /tf published along is odom->base_link.

So, I tried with my robot odom and the demo_turtlebot_tango.launch without making any editions to the original file. I also installed the turtlebot navigation package. Both in mapping and localization mode and I get the following error:

`[ WARN] [1524122911.508978143]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmapviz subscribed to (exact sync):
   /tango/camera/color_1/image_raw/relay,
   /tango/registered_depth,
   /tango/camera/color_1/camera_info,
   /scan
`
When I do a rostopic for each of these topics in the warning, they all seem to be published fine. Would you have any idea why this would happen ? 

Alternately, I use the following commands in different terminal windows:

- `$ rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/tango/point_cloud camera_info:=/tango/camera/color_1/camera_info _fixed_frame_id:=odom _decimation:=2 _fill_holes_size:=5`

- A separate launch file with the following code (I still need to update transforms for my robot): 
`<launch>`
` <arg name="tango_odometry"    default="true"/>`
`<node if="$(arg tango_odometry)" pkg="tf" type="static_transform_publisher" name="device_base_link" args="-0.3 0 0 0 1.5707963267948966 0 device base_footprint 100" />`
`<node if="$(arg tango_odometry)" pkg="tf" type="static_transform_publisher" name="odom_start_of_service_link" args="0 0 0.3 0 0 0 odom start_of_service 100" />`
``
`<node name="scan_relay" type="relay" pkg="topic_tools" args="/tango/laser_scan /scan"/>`
`<include file="$(find turtlebot_navigation)/launch/includes/move_base.launch.xml"/> `
``
`<node name="republish_rgb" type="republish" pkg="image_transport" args="compressed in:=/tango/camera/color_1/image_raw raw out:=/tango/camera/color_1/image_raw/relay"`
`</launch>`

- Mapping -->`$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start --Mem/ImagePreDecimation 2 --Mem/ImagePostDecimation 2 --Rtabmap/DetectionRate 0 --Kp/DetectorStrategy 6 --Vis/EstimationType 1" visual_odometry:=false odom_frame_id:="odom" frame_id:="base_footprint" rgb_topic:=/tango/camera/color_1/image_raw depth_topic:=/image camera_info_topic:=/tango/camera/color_1/camera_info compressed:=true depth_image_transport:=raw approx_sync:=false wait_for_transform:=0.2 `

- Localization --> `roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--Mem/ImagePreDecimation 2 --Mem/ImagePostDecimation 2 --Rtabmap/DetectionRate 0 --Kp/DetectorStrategy 6 --Vis/EstimationType 1" localization:=true visual_odometry:=false odom_frame_id:="odom" frame_id:="base_footprint" rgb_topic:=/tango/camera/color_1/image_raw depth_topic:=/image camera_info_topic:=/tango/camera/color_1/camera_info compressed:=true depth_image_transport:=raw approx_sync:=false wait_for_transform:=0.2 `

But in this case rtabmap.launch just freezes even on a rostopic query for /rtabmap/grigmap. the snapshot of the terminals is attached below:

[image]

Best Regards,
PG

matlabbe, 2018-04-19:
At least, are you able to teleop the robot by sending commands on a /cmd_vel topic (which is a [geometry_msgs/Twist](http://docs.ros.org/api/geometry_msgs/html/msg/Twist.html) msg)? I may suggest to get basic navigation running before integrating with rtabmap. See http://wiki.ros.org/navigation/Tutorials/RobotSetup

fiml, 2018-04-19:
Yea I am able to teleoperate. That works fine.

matlabbe, 2018-04-19:
Like the turtlebot-tango tutorial, move_base should be able to get proper TF to get the current pose of the robot (base_link or base_footprint depending of the configuration) in /map or /odom frame and the /map from rtabmap (a remap from /rtabmap/grid_map) to be able to plan a global plan.

You can debug the local and global plans of move_base by showing the corresponding paths in RVIZ. If both paths are published when you send a goal to move_base, verify that cmd_vel is also sent by move_base, if so make sure to connect it to input cmd_vel of your robot.

cheers,
Mathieu