introlab/rtabmap #281: The problem with rtabmap-console

Gatsby23, 2018-05-08:


matlabbe, 2018-05-08:
Do you have the full log of gdb with "`bt`"? What is your system and rtabmap version?

Gatsby23, 2018-05-09:
The version of my rtabmap is 0.17, and I can get the LogF and LogI files under the GUI model. I'm sorry that I don't know how to get the full log of gdb with "bt", could you please provide me some blogs?

pet_wolf23@163.com
 
From: matlabbe
Date: 2018-05-08 23:54
To: introlab/rtabmap
CC: Gatsby23; Author
Subject: Re: [introlab/rtabmap] The problem with rtabmap-console (#281)
Do you have the full log of gdb with "bt"? What is your system and rtabmap version?

matlabbe, 2018-05-10:
Example:
```bash
$ gdb rtabmap-console ...
[gdb] $ run
... wait until crash
[gdb] $ bt
```

Copy full log here.

cheers,
Mathieu