introlab/rtabmap #287: ORB2 SLAM

AnGPU, 2018-06-13:
Hi Matteu
after several attempts following this http://official-rtab-map-forum.67519.x6.nabble.com/How-to-use-ORB-SLAM2-library-in-rtabmap-td4230.html I managed to get the orb2 slam to be recognized in the cmake but after continuing the installation, trying to run rtabmap to set the odometry strategy I got this before anything works:
"rtabmap: error while loading shared libraries: libg2o.so: cannot open shared object file: No such file or directory" but it is in the place pointed by the  findORB_SLAM2.cmake 
I found almost similar situation with another libraries here and try sudo ldconfig but didn't work.

matlabbe, 2018-06-18:
Hi,

You should not have to modify the cmake files. When using ORB_SLAM2, you cannot link on g2o installed in /usr/local. When doing "cmake WITH_G2O=OFF WITH_ORB_SLAM2=ON ..", g2o version in ORB_SLAM2 (which is not compatible with g2o version installed by source) will be used instead. On runtime, you should make sure to add the path of g2o library built in ORB_SLAM2 library to LD_LIBRARY_PATH so that libg2o.so can be found:

`
rtabmap: error while loading shared libraries: libg2o.so: cannot open shared object file: No such file or directory
`

This is what I do in my .bashrc file:
`$ export LD_LIBRARY_PATH=/home/mathieu/workspace/ORB_SLAM2/lib:/home/mathieu/workspace/ORB_SLAM2/Thirdparty/g2o/lib:/home/mathieu/workspace/ORB_SLAM2/Thirdparty/DBoW2/lib:$LD_LIBRARY_PATH
`

cheers,
Mathieu

AnGPU, 2018-06-19:
Thanks! 
Now it runs but sends an Error message related with this part of the OdometryORBSLAM2.cpp code.
	"if(data.imageRaw().empty() ||
		data.imageRaw().rows != data.depthOrRightRaw().rows ||
		data.imageRaw().cols != data.depthOrRightRaw().cols)
	{
		UERROR("Not supported input!");
		return t;
	}"
"[ERROR] (2018-06-19 11:40:53.312) OdometryORBSLAM2.cpp:862::computeTransform() Not supported input!"
Should I open another issue and close this or ask here directly? I supposed it could be a problem from the source but I've never seen this kind of problem working with the configuration of the realsense zr300.

matlabbe, 2018-06-19:
In the commit above, I added an option to RealSense driver to scale depth image to RGB size so that both images have the same size, which is required for ORB_SLAM2. The option is in Preferences->Source, then under RealSense group box: "Depth image scaled to RGB image size.". You can give a try.

AnGPU, 2018-06-20:
It runs for several seconds and then segmentation fault. 
trying with gdb shows this:

Thread 21 "rtabmap" received signal SIGSEGV, Segmentation fault.
[Changing to Thread 0x7fff92003700 (LWP 9958)]
0x00007fffe70a7a15 in ORB_SLAM2::Optimizer::PoseOptimization(ORB_SLAM2::Frame*)
    () from /home/antonioguerrero/ORB_SLAM2/lib/libORB_SLAM2.so

matlabbe, 2018-06-20:
You can do "bt" in gdb to get a backtrace. However, the error seems coming from ORB_SLAM2 library. I don't have this error on my system though, I can run it for minutes without crashing. Can you show how did you build ORB_SLAM2 and rtabmap (cmake command lines and cmake status)? Are you using PCL 1.8 as it may be an Eigen problem (PCL 1.8 is built with `-march=native`)? Can you also show output of `ldd rtabmap`? 

cheers,
Mathieu

matlabbe, 2018-06-28:
I just checked the differences in my clone of ORB_SLAM2 code, and I removed` -march=native` flags:

```
~/workspace/ORB_SLAM2$ git diff Thirdparty/g2o/CMakeLists.txt
diff --git a/Thirdparty/g2o/CMakeLists.txt b/Thirdparty/g2o/CMakeLists.txt
index 620e688..ba8396c 100644
--- a/Thirdparty/g2o/CMakeLists.txt
+++ b/Thirdparty/g2o/CMakeLists.txt
@@ -54,8 +54,8 @@ IF(OPENMP_FOUND AND G2O_USE_OPENMP)
 ENDIF(OPENMP_FOUND AND G2O_USE_OPENMP)
 
 # Compiler specific options for gcc
-SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3 -march=native") 
-SET(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 -march=native") 
+SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} ") 
+SET(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE}") 
 
 # activate warnings !!!
 SET(g2o_C_FLAGS "${g2o_C_FLAGS} -Wall -W")

~/workspace/ORB_SLAM2$ git diff Thirdparty/DBoW2/CMakeLists.txt
diff --git a/Thirdparty/DBoW2/CMakeLists.txt b/Thirdparty/DBoW2/CMakeLists.txt
index 0eb5126..db34af2 100644
--- a/Thirdparty/DBoW2/CMakeLists.txt
+++ b/Thirdparty/DBoW2/CMakeLists.txt
@@ -1,8 +1,8 @@
 cmake_minimum_required(VERSION 2.8)
 project(DBoW2)
 
-set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3 -march=native ")
-set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall  -O3 -march=native")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall ")
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall ")
 
 set(HDRS_DBOW2
   DBoW2/BowVector.h

~/workspace/ORB_SLAM2$ git diff CMakeLists.txt
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 08a8af4..8190289 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -7,8 +7,8 @@ ENDIF()
 
 MESSAGE("Build type: " ${CMAKE_BUILD_TYPE})
 
-set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3 -march=native ")
-set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall   -O3 -march=native")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  ")
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall   ")
 
 # Check C++11 or C++0x support
 include(CheckCXXCompilerFlag)
```
You may give a try.

AnGPU, 2018-07-04:
Thanks
That seemed to be the problem and now I can run it perfectly.

AnGPU, 2018-07-16:
trying to reinstall the orb slam2 function, after searching every file from rtabmap and remove it I can't because of this errors at the end of the make -j4

[image]

in the process modifying the files from the comitt with this steps
git clone https://github.com/introlab/rtabmap.git rtabmap
cd rtabmap/build
corelib/include/rtabmap/core/CameraRGBD.h
corelib/src/CameraRGBD.cpp
corelib/src/OdometryORBSLAM2.cpp
guilib/src/PreferencesDialog.cpp
guilib/src/ui/preferencesDialog.ui 
cmake -DWITH_G2O=OFF ..
make -j4
sudo make install

matlabbe, 2018-07-16:
It is strange, these functions should be in PreferencesDialog.cpp: 
https://github.com/introlab/rtabmap/blob/89a0eb506b7e45464d76248bf778bf13646fc5cb/guilib/src/PreferencesDialog.cpp#L5338-L5341
https://github.com/introlab/rtabmap/blob/89a0eb506b7e45464d76248bf778bf13646fc5cb/guilib/src/PreferencesDialog.cpp#L4416-L4419

AnGPU, 2018-07-17:
I don't know. I don't remember to change more than what was made in the commit and don't open it again after that. 
Anyway I copy the file again from here and now it works. Thanks for pointing the file.

AnGPU, 2018-07-20:
I'm working in the way to add the commit raulmur/ORB_SLAM2#21 to reduce significantly the time of vocabulary reading within a binary file instead of the txt one which is determining to test this into a embedded system.  So I will let this issue open if theres's no inconvenience.

shenlan1994, 2019-12-02:
I tried to compile rtab-map with orb-slam2, but a segmentation fault occurred. The gdb debugging information is as follows:
Thread 35 "rtabmap" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff48898700 (LWP 23066)]
0x00007ffff71c4430 in g2o::BaseBinaryEdge<6, g2o::SE3Quat, g2o::VertexSE3Expmap, g2o::VertexSE3Expmap>::linearizeOplus() ()
   from /home/tao/Documents/RTAB-Map/rtabmap/bin/librtabmap_core.so.0.19
(gdb) BT
#0  0x00007ffff71c4430 in g2o::BaseBinaryEdge<6, g2o::SE3Quat, g2o::VertexSE3Expmap, g2o::VertexSE3Expmap>::linearizeOplus() ()
   from /home/tao/Documents/RTAB-Map/rtabmap/bin/librtabmap_core.so.0.19
#1  0x00007ffff71c340c in g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::buildSystem()
    () from /home/tao/Documents/RTAB-Map/rtabmap/bin/librtabmap_core.so.0.19
#2  0x00007fffea747068 in g2o::OptimizationAlgorithmLevenberg::solve(int, bool) ()
   from /home/tao/slam/ORB_SLAM2/Thirdparty/g2o/lib/libg2o.so
#3  0x00007fffea73e3e7 in g2o::SparseOptimizer::optimize(int, bool) ()
   from /home/tao/slam/ORB_SLAM2/Thirdparty/g2o/lib/libg2o.so
#4  0x00007ffff71bafa8 in rtabmap::OptimizerG2O::optimizeBA(int, std::map<int, rtabmap::Transform, std::less<int>, std::allocator<std::pair<int const, rtabmap::Transform> > > const&, std::multimap<int, rtabmap::Link, std::less<int>, std::allocator<std::pair<int const, rtabmap::Link> > > const&, std::map<int, rtabmap::CameraModel, std::less<int>, std::allocator<std::pair<int const, rtabmap::CameraModel> > > const&, std::map<int, cv::Point3_<float>, std::less<int>, std::allocator<std::pair<int const, cv::Point3_<float> > > >&, std::map<int, std::map<int, rtabmap::FeatureBA, std::less<int>, std::allocator<std::pair<int const, rtabmap::FeatureBA> > >, std::less<int>, std::allocator<std::pair<int const, std::map<int, rtabmap::FeatureBA, std::less<int>, std::allocator<std::pair<int const, rtabmap::FeatureBA> > > > > > const&, std::set<int, std::less<int>, std::allocator<int> >*) ()
   from /home/tao/Documents/RTAB-Map/rtabmap/bin/librtabmap_core.so.0.19
#5  0x00007ffff7272290 in rtabmap::OdometryF2M::computeTransform(rtabmap::SensorData&, rtabmap::Transform const&, rtabmap::OdometryInfo*) ()
   from /home/tao/Documents/RTAB-Map/rtabmap/bin/librtabmap_core.so.0.19

 I don't know where to find the problem

matlabbe, 2019-12-13:
There is a patch [here](https://github.com/introlab/rtabmap/blob/af9e11311275a9cf0ac9a9294d3a167808fb0e72/docker/jfr2018/Dockerfile#L139) to make ORB_SLAM2 without `-mach=native` flag. Your error maybe related to this flag, as g2o is using Eigen.