introlab/rtabmap #290: cannot get picture?

GeorgeTower, 2018-06-22:
when i use the follow C++,   a problem turn out,

    odomThread->registerToEventsManager();
    rtabmapThread->registerToEventsManager();
    mapBuilder->registerToEventsManager();
    UEventsManager::createPipe(cameraThread, odomThread, "CameraEvent");
    UEventsManager::createPipe(odomThread, rtabmapThread, "OdometryThread");
    rtabmapThread->start();
    odomThread->start();
    cameraThread->start();
    mapBuilder->show();

there is no picture i can get.  i cannot find the reason!   please give me a hand! thank u very much!
[image]

GeorgeTower, 2018-06-22:
using the UsbDk-1.00-19, and the initial is well, means i can get link with the kinectv2 , but when read, using the above program, got nothing!    i think is the reason is UsbDk-1.00-19! but i can not sure!

GeorgeTower, 2018-06-22:
windows is  win10,  devices is surface pro 3! thanks  a lot!

matlabbe, 2018-06-28:
From the Windows instructions, UsbDK v1.00-12 is used: https://github.com/introlab/rtabmap/wiki/Installation#windows

If you are building rtabmap from source, it is also possible using Kinect4Windows SDK instead of freenect2.