introlab/rtabmap #300: what is maxscale in calibration ui in rtabmap

pras2020, 2018-08-02:
what is maxscale in calibration ui in rtabmap . what is the advantage and disadvantage of maxscale. what is the best value.

matlabbe, 2018-08-03:
The max scale is used here:
https://github.com/introlab/rtabmap/blob/f938e8ce2902bdf60630d988f225f302ba3689a9/guilib/src/CalibrationDialog.cpp#L390-L408
The image is scaled down until "max scale" to help finding the board. This can be useful for high definition images. For example, for 1080p images like Kinect v2, the max scale can be increased (to 2 or 3) to help finding the chessboard.

pras2020, 2018-08-13:
Is the above comments mean that if we go for higher resolution stereo images we need to increase the maxscale from 1 to 2 or 3.

matlabbe, 2018-08-14:
Yes, though it is optional. If the checkboard is always detected with max scale 1, you can keep it that way.