introlab/rtabmap #302: reserved identifier violation

elfring, 2018-08-06:
I would like to point out that an identifier like “[`_DepthSpacePoint`](https://github.com/introlab/rtabmap/blob/35bc2d06a6e95f3ebfe7011a8a743fa74fef4655/corelib/include/rtabmap/core/CameraRGBD.h#L96 "Update candidate")” [does not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

matlabbe, 2018-08-07:
Hi,
That variable is coming from the official Microsoft Kienct for Windows SDK:
c:\Program Files\Microsoft SDKs\Kinect\v2.0_1409\inc\Kinect.h
```cpp
#ifndef _DepthSpacePoint_
#define _DepthSpacePoint_
typedef struct _DepthSpacePoint
    {
    float X;
    float Y;
    } 	DepthSpacePoint;

#endif // _DepthSpacePoint_
```

There is a compilation error if we do this instead in CameraRGBD.h for the forward declaration of the structure:
```cpp
typedef struct DepthSpacePoint DepthSpacePoint;
```

If you have ideas on how the forward declaration should be done to avoid the underscore prefix, I'm open for suggestions.

elfring, 2018-08-08:
How do you think about to [inform any other software designers (and corresponding maintainers) about a questionable identifier selection](https://developer.microsoft.com/en-us/windows/kinect "Kinect for Windows")?

matlabbe, 2018-08-09:
For **optional** third party libraries, I don't spend too much time reporting/fixing/follow up their bugs unless they are breaking rtabmap functionalities or they are very obvious. For the underscore prefix used by Microsoft in this header file, this may satisfy their own coding standard. At least if the code is consistent, I don't think it is a problem.

matlabbe, 2018-08-14:
We try to fix as many compilation warnings as we can, though some often are coming from third party libraries. As long as they don't break the functionality, they are low priority issues.