introlab/rtabmap #303: Build successfully on Windows, but crash when view point clouds.

ask8655ok, 2018-08-08:
Hi everyone,
I have successfully built Rtabmap from the source code on Windows 10.
However, after I used xtion pro/ xtion 2/ realsense d435 to scan something, I would like to view the point cloud with mesh and texture. So I clicked [Edit]->[View point clouds...]->check [Meshing]->check [Texture mapping], and waited for the processing.
And Rtabmap crashed.
[image]

VS shown that Rtabmap crashed at rtabmap_gui.dll, and that the unhandled exception condition was 0xC0000005.

The following is my building environment, 

CMake 3.12.0, Qt 5.11.1, Visual Studio 2017
Rtabmap source code version: 0.17.3. (Version 0.16.3 crashed as well.)

Info :
  Version : 0.17.3
  CMAKE_INSTALL_PREFIX = C:/Program Files/RTABMap
  CMAKE_BUILD_TYPE =     
  CMAKE_INSTALL_LIBDIR = lib
  BUILD_APP =            ON
  BUILD_TOOLS =          ON
  BUILD_EXAMPLES =       ON
  CMAKE_CXX_FLAGS = /DWIN32 /D_WINDOWS /W3 /GR /EHsc  -openmp
  PCL_DEFINITIONS =   -D__SSSE3__ -D__SSE2__ -D__SSE__;-DBOOST_ALL_NO_LIB;-DDISABLE_OPENNI;-DDISABLE_ENSENSO;-DDISABLE_DAVIDSDK;-DDISABLE_DSSDK;-DDISABLE_RSSDK;-DDISABLE_PCAP;-DDISABLE_PNG;-DFLANN_STATIC
  With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = NO (not found, License: BSD)
  With Freenect             = NO (libfreenect not found)
  With OpenNI2              = YES (License: Apache v2)
  With Freenect2            = NO (libfreenect2 not found)
  With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
  With dc1394               = NO (dc1394 not found)
  With FlyCapture2/Triclops = NO (Point Grey SDK not found)
  With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
  With g2o                  = YES (License: BSD)
  With GTSAM                = NO (GTSAM not found)
  With VERTIGO              = YES (License: GPLv3)
  With cvsba                = NO (cvsba not found)
  With libpointmatcher      = NO (WITH_POINTMATCHER=OFF)
  With loam_velodyne        = NO (WITH_LOAM=OFF)
  With ZED                  = NO (ZED sdk not found)
  With RealSense            = NO (librealsense not found)
  With RealSense2           = YES (License: Apache-2)
  With OCTOMAP              = YES (License: BSD)
  With CPUTSDF              = NO (CPUTSDF not found)
  With OpenChisel           = NO (open_chisel not found)
  With libfovis             = NO (libfovis not found)
  With libviso2             = NO (libviso2 not found)
  With dvo_core             = NO (dvo_core not found)
  With okvis                = NO (okvis not found)
  With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
  With Qt5                  = YES (License: Open Source or Commercial)

Please help me solve this problem. I'll appreciate you a lot!!
//btw, do I have a pleasure of using the released version 0.17.3? I'll appreciate that a lot, too!!

ask8655ok, 2018-08-09:
The version of VTK I used is 8.0.
I downloaded "PCL 1.8.1" and there were many libraries contained in the folder "3rdParty", like Boost, Eigen, VTK, etc. Should I download these libraries individually?
Besides, I have tried the artifact(0.17.4). It works! Then I read the dependencies lists, and it seems that the version of VTK the artifact(0.17.4) used is 6_3. Does the difference of the two versions cause the crashing problem?
Thank you for replying me!!

matlabbe, 2018-08-09:
I know that with VTK8 on Ubuntu, the PCL visualizer cannot show a mesh with texture, though it should not crash (mesh will be only shown in white). On Windows, it may crash. I had a lot of problems on Windows to get all libraries at the right versions so that all visualization functionalities work. I cannot help more than that. Unless you need a specific version of a third party library, you may download each one linked in that [file](https://github.com/introlab/rtabmap/blob/master/.appveyor.yml).

ask8655ok, 2018-08-10:
After I replaced all the libraries I used with those mentioned in your dependency list(except for Qt(5.11.1), OpenNI2(Asus Xtion 2 version), RealSense2, GTSAM(I failed...) ), Rtabmap runs successfully and peacefully.
Thanks!!

But, there is another problem here...
This morning I
> included gtsam, 
> set GTSAM_DIR to 'C:/Program Files/GTSAM/CMake', 
> set GTSAM_INCLUDE_DIR to 'C:/Program Files/GTSAM/include', 
> and set GTSAM_LIB to 'C:/Program Files/GTSAM/lib/libgtsam.lib',

a linker error came up.
The error msg was
> LINK : fatal error LNK1181: cannot open input file 'gtsam.lib' (in rtabmap_core.vcxproj)

I could only find libgtsam.lib so I tried to set the path 'gtsam.lib' to 'C:/Program Files/GTSAM/lib/libgtsam.lib' in [rtabmap_core.vcxproj's Project properties]->[linker]->[input]->[Additional Dependencies].

Then the error msg became lots of different error msgs, like
> libgtsam.lib(LevenbergMarquardtOptimizer.obj) : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl boost
::timer::cpu_timer::cpu_timer(void)" (__imp_??0cpu_timer@timer@boost@@QEAA@XZ) referenced in function "public: bool __cdecl gtsam::Levenbe
rgMarquardtOptimizer::tryLambda(class gtsam::GaussianFactorGraph const &,class gtsam::VectorValues const &)" (?tryLambd
a@LevenbergMarquardtOptimizer@gtsam@@QEAA_NAEBVGaussianFactorGraph@2@AEBVVectorValues@2@@Z)

Have I done something wrong? Or this error on Windows is unavoidable....

matlabbe, 2018-08-10:
It seems the boost version is not the same. If you built GTSAM yourself, see step 8 [here](https://github.com/introlab/rtabmap/wiki/Installation#source-1). There is a patch to apply and GTSAM library should be built as static to avoid some errors.