introlab/rtabmap #304: Build with msckf_vio results on several errors

TSC21, 2018-08-12:
Hi @matlabbe,

Setting the build with `-DWITH_MSCKF_VIO=ON` set results on the following errors (sent in a pastebin link since it's a big log): https://pastebin.com/yamP0ktE. Seems all related to the usage of private members. Anything you can think of?

Thanks in advance!

matlabbe, 2018-08-12:
To build with msckf, we should remove ros stuff from their library and put other stuff as public. See this patch: https://gist.github.com/matlabbe/f2518d7427e7f6740af2110e540b1f2b

TSC21, 2018-08-12:
> To build with msckf, we should remove ros stuff from their library and put other stuff as public. See this patch: https://gist.github.com/matlabbe/f2518d7427e7f6740af2110e540b1f2b

Do you have that available in a fork of yours? Also, what's the performance of the VO compared to the other frameworks you have an integration with?

TSC21, 2018-08-13:
Well never mind. I guess I can patch it myself. Thanks for sharing the patch. Would be good to understand though if there's a clear improvement compared with the other available VO improvements.

matlabbe, 2018-08-14:
We have a paper under review showing some comparisons between available odometry approaches in RTAB-Map.  Here is a snippet evaluating [MSCKF_VIO](https://github.com/KumarRobotics/msckf_vio) on [EuRoC](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets) dataset:
[image]

MSCKF is the fastest approach, though not necessarily giving the best accuracy.

TSC21, 2018-08-14:
> MSCKF is the fastest approach, though not necessarily giving the best accuracy.

Thanks for the above. Though, it also seems to offer an interesting balance between accuracy and speed as well.

matlabbe, 2018-08-14:
Yes, the speed is a huge advantage, in particular on systems with low computing resources.

shirbarzel, 2020-07-09:
> To build with msckf, we should remove ros stuff from their library and put other stuff as public. See this patch: https://gist.github.com/matlabbe/f2518d7427e7f6740af2110e540b1f2b

Hi I tried your patch on the msckf from kumar robotics:

```
git clone https://github.com/KumarRobotics/msckf_vio ~/catkin_ws/src/
cd ~/catkin_ws/src/msckf_vio/
git apply msckf_vio_a9386c5_ros_commented.patch
```

But I got:
error: corrupt patch at line 637

Is it aligned with kumar robotics master branch? If not from where I can get the fork you did?

matlabbe, 2020-07-11:
Not sure what are your setup. You may try this [example](https://github.com/introlab/rtabmap_ros/blob/34c75139f253b086d4f4f9f94776e7b7ec059e34/launch/tests/euroc_datasets.launch#L13-L19) to make sure it is working. I didn't test more than with EuRoC rosbag, but I assume if your camera gives similar topics, it would be okay.