introlab/rtabmap #306: RTABMap with MSCKF_VIO: odometry quality always 0

TSC21, 2018-08-14:
Hi @matlabbe,

I applied the patch you gave in https://github.com/introlab/rtabmap/issues/304 and recompiled rtabmap with it. On my launch file I also remapped the IMU topic to the topic that's publishing the IMU data. But, after launching the visual odometry plus rtabmap on the same node, I get the odometry quality always 0:
```bash
[ INFO] [1534245743.357730936, 1528193211.151827867]: Odom: quality=0, std dev=0.095018m|0.015811rad, update time=0.059044s
[ INFO] [1534245743.575522027, 1528193211.373813285]: Odom: quality=0, std dev=0.096536m|0.015811rad, update time=0.216898s
[ INFO] [1534245743.695364286, 1528193211.495279104]: Odom: quality=0, std dev=0.098518m|0.015811rad, update time=0.118965s
[ INFO] [1534245743.947517081, 1528193211.746689131]: Odom: quality=0, std dev=0.102831m|0.015811rad, update time=0.250936s
[ INFO] [1534245744.067265414, 1528193211.867914081]: Odom: quality=0, std dev=0.101889m|0.015811rad, update time=0.118922s
[ INFO] [1534245744.155808905, 1528193211.948482027]: Odom: quality=0, std dev=0.106454m|0.015811rad, update time=0.087117s
[ INFO] [1534245744.192022434, 1528193211.988677775]: Odom: quality=0, std dev=0.099700m|0.015811rad, update time=0.035235s
[ INFO] [1534245744.254966478, 1528193212.049161587]: Odom: quality=0, std dev=0.099799m|0.015811rad, update time=0.061826s
[ INFO] [1534245744.334768848, 1528193212.129676880]: Odom: quality=0, std dev=0.099003m|0.015811rad, update time=0.079005s
[ INFO] [1534245744.423181892, 1528193212.220448279]: Odom: quality=0, std dev=0.099499m|0.015811rad, update time=0.087588s
[ INFO] [1534245744.499913788, 1528193212.300999837]: Odom: quality=0, std dev=0.098895m|0.015811rad, update time=0.075781s
[ INFO] [1534245744.712627273, 1528193212.512384034]: Odom: quality=0, std dev=0.100240m|0.015811rad, update time=0.211706s
[ INFO] [1534245745.092937692, 1528193212.885267934]: Odom: quality=0, std dev=0.101742m|0.015811rad, update time=0.379422s
[ INFO] [1534245745.392717595, 1528193213.187449558]: Odom: quality=0, std dev=0.108091m|0.015811rad, update time=0.297606s
[ INFO] [1534245745.432299600, 1528193213.227716228]: Odom: quality=0, std dev=0.109623m|0.015811rad, update time=0.038616s
[ INFO] [1534245745.512160720, 1528193213.308209381]: Odom: quality=0, std dev=0.119842m|0.015811rad, update time=0.078502s
[ INFO] [1534245745.577312821, 1528193213.378658296]: Odom: quality=0, std dev=0.123543m|0.015811rad, update time=0.062476s
[ INFO] [1534245745.659355306, 1528193213.459258275]: Odom: quality=0, std dev=0.126773m|0.015811rad, update time=0.077773s
[ INFO] [1534245745.691593761, 1528193213.489443435]: Odom: quality=0, std dev=0.128670m|0.015811rad, update time=0.031514s
[ INFO] [1534245745.778745485, 1528193213.580709293]: Odom: quality=0, std dev=0.130663m|0.015811rad, update time=0.086131s
[ INFO] [1534245746.222166858, 1528193214.023699469]: Odom: quality=0, std dev=0.129807m|0.015811rad, update time=0.442685s
```

The data that I am playing is from a rosbag, with `use_sim_time` set. Any suggestion on how to debug this or what's causing this? I have all the MSCKF parameters from RTABMap visual odometry set to default.

Thanks in advance!

TSC21, 2018-08-14:
Also a curious thing above is that it seems that the std deviation of the position is being updated (so I suppose that the msckf is working?), but the attitude doesn't. Maybe something is missing on the approach?

matlabbe, 2018-08-14:
As long at it doesn't say "lost", the odometry is working. You should be able to show `$ rostopic echo /rtabmap/odom` and see that values are computed. Those outputs are default ones for all odometry approaches. If for a third party approach we don't have access (or it is not implemented) to the number of inliers used for odometry, quality will be always 0. MSCKF library doesn't compute values for angular twist covariance, so they are [fixed](https://github.com/introlab/rtabmap/blob/master/corelib/src/OdometryMSCKF.cpp#L950) accordingly to gyro noise parameter.

Note however that MSCKF has only being tested on EuRoC datasets, and covariance values were tuned to get good results on graph optimizations. Not sure if the same tuning can apply on other VI sensors.

TSC21, 2018-08-14:
> As long at it doesn't say "lost", the odometry is working. You should be able to show $ rostopic echo /rtabmap/odom and see that values are computed.

I actually did it and no output is given...

TSC21, 2018-08-14:
Also I am not able to launch `rtabmapviz` after building rtabmap agains msckf_vio, immediately dropping a segfault. Probably better to open a new issue with the gdb dump?

matlabbe, 2018-08-14:
I updated the launch file [euroc_datasets.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/euroc_datasets.launch) for some examples. You can download the corresponding rosbags on their [website](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets#downloads). There is an example for MSCKF:
```bash
$ roslaunch rtabmap_ros euroc_datasets.launch args:="-d RGBD/CreateOccupancyGrid false Odom/Strategy 8"
$ rosbag play --clock V1_01_easy.bag
```

To debug the rtabmapviz seg fault, the best would be to show the cmake output of rtabmap library, and also the backtrace with [gdb](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB). There is maybe a mismatch between libraries used to build the library and those actually used on runtime. This can be also an Eigen problem.