introlab/rtabmap #307: link rtabmap to a new  library

zoumaguanxin, 2018-09-12:
Hi,
I intend to add a new module to rtabmap as an alternative icp algorithm. Conforming to  your software framework, I make the module optional. But when I revised your CMakeLists.txt in top directory, I cannot find  the configuration of generating targets like the following code.
```
add_library(rtabmap_core ${RTBAMAP_CORE_SRC})
target_link_libraries(rtabmap_core ${CONF_DEPENDENCIES}
```
Therefore, I failed to link them to my icp library. I am not good at CMake programming, can you tell me how you configure the related target files?

thanks,
Shawn

zoumaguanxin, 2018-09-13:
I found it in  corelib/src/rtabmap_core. There are lots of cpp files so that I ignored it

closed 2018-09-13 by zoumaguanxin

matlabbe, 2018-09-13:
For the record:
https://github.com/introlab/rtabmap/blob/5f6dd0846d9c63ae8d624c4230e16f42d93fbf20/corelib/src/CMakeLists.txt#L500-L501