introlab/rtabmap #308: 0.17.5 and Realsense sdk 2 with d415 on win10

TheBricktop, 2018-09-12:
Hey hey people,
Id' like to use realsense d415 to scan and map an architectural object, 
but there is no other free software to use than rtab map, so i've downloaded 17.5 nightly that supposed to have realsense2 support but after installing every driver for d415 and sdk, in RTABMAP any sources other than kinect1 are greyed out, how can I install additional grabbers for realsense2 or (at least) for kinect 2.
Thanks for any help.

matlabbe, 2018-09-14:
I just released the 0.17.5 version with RealSense 2 driver. You can download it here: https://github.com/introlab/rtabmap/releases/tag/0.17.5

The driver for D435 should work for D415, unless there is an API difference between both. I can only test here for D435. Let me know if it is working with your D415, then I will update the UI to add D415 in the menu.

cheers,
Mathieu

TheBricktop, 2018-09-14:
Thank You!
Ill let You know if it is working properly.
Cheers,
Maciej

TheBricktop, 2018-10-30:
It's working very well with d415. Closing

closed 2018-10-30 by TheBricktop