introlab/rtabmap #309: Build error on Ubuntu 16.04

alexsmartens, 2018-09-14:
```
$:~/Programs/rtabmap/build$ make 
Scanning dependencies of target rtabmap_utilite
[  0%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsManager.cpp.o
[  1%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsHandler.cpp.o
[  1%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsSender.cpp.o
[  1%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UFile.cpp.o
[  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UDirectory.cpp.o
[  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UConversion.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/ULogger.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UThread.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UTimer.cpp.o
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UProcessInfo.cpp.o
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UVariant.cpp.o
[  4%] Linking CXX shared library ../../../bin/librtabmap_utilite.so
[  4%] Built target rtabmap_utilite
Scanning dependencies of target res_tool
[  5%] Building CXX object utilite/resource_generator/CMakeFiles/res_tool.dir/main.cpp.o
[  5%] Linking CXX executable ../../../bin/rtabmap-res_tool
[  5%] Built target res_tool
[  5%] [Creating resources]
 Using namespace=rtabmap
 Using target directory=/home/blaz/Programs/rtabmap/build/corelib/src
Input file "/home/blaz/Programs/rtabmap/corelib/src/resources/DatabaseSchema.sql" size = 5431 bytes
Output file "/home/blaz/Programs/rtabmap/build/corelib/src/DatabaseSchema_sql.h" size = 11089 bytes
Scanning dependencies of target rtabmap_core
[  5%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RtabmapThread.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Statistics.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o
/home/blaz/Programs/rtabmap/corelib/src/Memory.cpp: In member function ‘rtabmap::Signature* rtabmap::Memory::createSignature(const rtabmap::SensorData&, const rtabmap::Transform&, rtabmap::Statistics*)’:
/home/blaz/Programs/rtabmap/corelib/src/Memory.cpp:3779:55: error: ‘CV_BGR2GRAY’ was not declared in this scope
     cv::cvtColor(decimatedData.imageRaw(), imageMono, CV_BGR2GRAY);
                                                       ^
/home/blaz/Programs/rtabmap/corelib/src/Memory.cpp:3882:46: error: ‘CV_BGR2GRAY’ was not declared in this scope
     cv::cvtColor(data.imageRaw(), imageMono, CV_BGR2GRAY);
                                              ^
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:139: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```
Any ideas?

matlabbe, 2018-09-15:
Which opencv version do you have? What is cmake output?

alexsmartens, 2018-09-15:
I am using OpenCV 4.0.0
```
$  pkg-config --modversion opencv
4.0.0
```

alexsmartens, 2018-09-15:
And here is cmake output:
```
$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found Eigen: /usr/include/eigen3  
-- Eigen found (include: /usr/include/eigen3, version: 3.3.4)
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Found libusb-1.0: /usr/include  
-- Checking for module 'flann'
--   Found flann, version 1.8.4
-- Found FLANN: /usr/lib/x86_64-linux-gnu/libflann_cpp.so  
-- FLANN found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp.so;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp.so)
-- FLANN found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp.so;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp.so)
-- Found Qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- Found PCL_COMMON: /usr/local/lib/libpcl_common.so  
-- Found PCL_OCTREE: /usr/local/lib/libpcl_octree.so  
-- Found PCL_IO: /usr/local/lib/libpcl_io.so  
-- Found PCL_KDTREE: /usr/local/lib/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/local/lib/libpcl_search.so  
-- Found PCL_SURFACE: /usr/local/lib/libpcl_surface.so  
-- Found PCL_FILTERS: /usr/local/lib/libpcl_filters.so  
-- Found PCL_2D: /usr/local/include/pcl-1.8  
-- Found PCL_FEATURES: /usr/local/lib/libpcl_features.so  
-- Found PCL_REGISTRATION: /usr/local/lib/libpcl_registration.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/local/lib/libpcl_sample_consensus.so  
-- Found PCL_GEOMETRY: /usr/local/include/pcl-1.8  
-- Found PCL_ML: /usr/local/lib/libpcl_ml.so  
-- Found PCL_SEGMENTATION: /usr/local/lib/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/local/lib/libpcl_visualization.so  
CMake Warning at CMakeLists.txt:183 (MESSAGE):
  PCL definitions contain "-march=native", make sure all libraries using
  Eigen are also compiled with that flag to avoid some segmentation faults
  (with gdb referring to some Eigen functions).

-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
-- Found OpenCV: /usr/local/include
-- Found PCL: /usr/local/include/pcl-1.8;/usr/include/eigen3;/usr/include;/usr/include/vtk-6.2;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/jsoncpp;/usr/include/x86_64-linux-gnu;/usr/include/hdf5/openmpi;/usr/include/python2.7;/usr/include/libxml2;/usr/include/tcl
-- Found ZLIB: /usr/include
[... 39 lines cut ...]
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--   With libpointmatcher      = NO (libpointmatcher not found)
--   With loam_velodyne        = NO (loam_velodyne not found)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With OCTOMAP              = NO (octomap not found)
--   With CPUTSDF              = NO (CPUTSDF not found)
--   With OpenChisel           = NO (open_chisel not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With okvis                = NO (okvis not found)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/Programs/rtabmap/build
```

picaro8, 2018-09-18:
I also have this issue, using OpenCV git trunk.

Since CV_BGR2GRAY is only defined at opencv2/imgproc/types_c.h, putting
#include "/usr/local/include/opencv2/imgproc/types_c.h"
at the beginning of Memory.cpp allows that file to compile, but other OpenCV errors occur elsewhere, e.g. CameraRGB.cpp.

And imgproc/types_c.h is included by imgproc/imgproc_c.h

However, the Memory.cpp etc modules include opencv2/imgproc/imgproc.hpp, where the enum is defined as COLOR_BGR2GRAY.

At any rate ended up including 
#include <opencv2/imgproc/imgproc_c.h>
#include <opencv2/videoio/videoio_c.h>

to all compilation units using any of the color enums.

Also many other similar changes. So not sure if this is a change in opencv or that some include was removed from a common header in rtabmap. Not sure what your philosophy is on bulk #including third party headers in internal headers, vs only in the .cpp file.

matlabbe, 2018-09-19:
RTAB-Map's code wasn't updated yet to OpenCV 4. They removed the includes of "c" headers from the "cpp" headers, breaking all codes depending on "c" definitions. We should explicitly add the "c" headers like in the first commit above to fix the problem. 

cheers,
Mathieu

closed 2018-09-19 by matlabbe