introlab/rtabmap #311: Opening the rtabmap.db file with **rtabmap-databaseViwer**,  error occurs while exporting the poses.txt

hewind1992, 2018-09-17:
[image]

As shown in the [picture-1](https://user-images.githubusercontent.com/35555593/45608322-cfd39700-ba84-11e8-9c1e-4095bfa511f3.png), the db file contains 932 frames. But the exported poses.txt file only contains **ONE** line. **How can I get the poses of all frames?**

[image]

---
Additional

I follow the page [MappingAndNavigationOnTurtlebot](http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot) to do the rtab mapping, and save the result as the db file.

And the development environment is the Ubuntu 16.04 with ROS1, plus turtlebot 2.

matlabbe, 2018-09-17:
Can you share the db? Or can you open the graph view in database viewer to see if there is a map?

matlabbe, 2018-09-20:
Hi,

By default, rtabmap adds all frames to database even if the robot is not moving (where corresponding nodes are not added to graph), for debugging purpose and ease to replay the database offline.

A new node is added to graph only if the robot has moved, based on those parameters (setting to 0 will make all frames added to graph):
```bash
$ rtabmap --params | grep RGBD/
Param: RGBD/LinearUpdate = "0.1" [Minimum linear displacement (m) to update the map. Rehearsal is done prior to this, so weights are still updated.]
Param: RGBD/AngularUpdate = "0.1" [Minimum angular displacement (rad) to update the map. Rehearsal is done prior to this, so weights are still updated.]
```
If you don't want to save frames to database when not moving, set parameter `Mem/NotLinkedNodesKept` to `false`:
```bash
$ rtabmap --params | grep NotLin
Param: Mem/NotLinkedNodesKept = "true" [Keep not linked nodes in db (rehearsed nodes and deleted nodes).]
```

For the single line pose, maybe the robot didn't move, so only the first frame has been added to graph.

cheers,
Mathieu