introlab/rtabmap #321: Installation problem for last version of RTAB-Map with loam_velodyne (not in ROS) in Ubuntu 16.04:loam::MultiScanMapper::getRingForAngle(float const&)Undefined references

Yeah2333, 2018-10-17:
Hello,when i run make.there are something wrong 
[ 83%] Built target rtabmap_gui
Scanning dependencies of target recovery
[ 83%] Building CXX object tools/Recovery/CMakeFiles/recovery.dir/main.cpp.o
[ 83%] Linking CXX executable ../../../bin/rtabmap-recovery
../../../bin/librtabmap_core.so.0.17.7：对‘loam::MultiScanMapper::getRingForAngle(float const&)’Undefined references
collect2: error: ld returned 1 exit status
tools/Recovery/CMakeFiles/recovery.dir/build.make:705: recipe for target '../bin/rtabmap-recovery' failed
make[2]: *** [../bin/rtabmap-recovery] Error 1
CMakeFiles/Makefile2:2364: recipe for target 'tools/Recovery/CMakeFiles/recovery.dir/all' failed

i edit CmakeLists.txt to turn off loam_velodyne support ,the question still exist.
option(WITH_CVSBA         "Include cvsba support"                ON)
option(WITH_POINTMATCHER  "Include libpointmatcher support"      ON)
option(WITH_LOAM          "Include LOAM support"                 OFF)
option(WITH_FLYCAPTURE2   "Include FlyCapture2/Triclops support" ON)

matlabbe, 2018-10-17:
Make sure you have this [pull request](https://github.com/laboshinl/loam_velodyne/pull/89). Note that it has been merged 4 hours ago in their master branch, so just update your `loam_velodyne` source and rebuild it.

Yeah2333, 2018-10-18:
The problem has been solved. Thank you very much.

closed 2018-10-18 by Yeah2333