introlab/rtabmap #322: Nvidia Jetson TX2 - installation error

Bogonek, 2018-10-21:
Hi everyone!

I'm trying to instal rtabmap on Nvidia Jetson TX2 developer board with Linux Ubuntu 16.04 on it.

During instalatrion I recive few errors:

`nvidia@tegra-ubuntu:~/rtabmap/build$ make -j4
[  5%] Built target rtabmap_utilite
[  6%] Built target res_tool
[  7%] Built target imagesJoiner
[  8%] Built target extractObject
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraImages.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenni.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRGBDImages.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenNICV.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp: In member function ‘virtual bool rtabmap::CameraOpenNICV::init(const string&, const string&)’:
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:59:23: error: ‘CV_CAP_OPENNI_ASUS’ was not declared in this scope
  _capture.open( _asus?CV_CAP_OPENNI_ASUS:CV_CAP_OPENNI );
                       ^
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:59:42: error: ‘CV_CAP_OPENNI’ was not declared in this scope
  _capture.open( _asus?CV_CAP_OPENNI_ASUS:CV_CAP_OPENNI );
                                          ^
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:62:17: error: ‘CV_CAP_OPENNI_IMAGE_GENERATOR_OUTPUT_MODE’ was not declared in this scope
   _capture.set( CV_CAP_OPENNI_IMAGE_GENERATOR_OUTPUT_MODE, CV_CAP_OPENNI_VGA_30HZ );
                 ^
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:62:60: error: ‘CV_CAP_OPENNI_VGA_30HZ’ was not declared in this scope
   _capture.set( CV_CAP_OPENNI_IMAGE_GENERATOR_OUTPUT_MODE, CV_CAP_OPENNI_VGA_30HZ );
                                                            ^
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:63:31: error: ‘CV_CAP_OPENNI_DEPTH_GENERATOR_FOCAL_LENGTH’ was not declared in this scope
   _depthFocal = _capture.get( CV_CAP_OPENNI_DEPTH_GENERATOR_FOCAL_LENGTH );
                               ^
In file included from /home/nvidia/rtabmap/corelib/src/../include/rtabmap/core/IMU.h:12:0,
                 from /home/nvidia/rtabmap/corelib/src/../include/rtabmap/core/SensorData.h:40,
                 from /home/nvidia/rtabmap/corelib/src/../include/rtabmap/core/Camera.h:33,
                 from /home/nvidia/rtabmap/corelib/src/../include/rtabmap/core/camera/CameraOpenNICV.h:32,
                 from /home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:27:
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:66:48: error: ‘CV_CAP_PROP_FRAME_WIDTH’ was not declared in this scope
   UINFO("FRAME_WIDTH        %f", _capture.get( CV_CAP_PROP_FRAME_WIDTH ));
                                                ^
/home/nvidia/rtabmap/utilite/include/rtabmap/utilite/ULogger.h:51:89: note: in definition of macro ‘ULOGGER_LOG’
 #define ULOGGER_LOG(level, ...) ULogger::write(level, __FILE__, __LINE__, __FUNCTION__, __VA_ARGS__)
                                                                                         ^
/home/nvidia/rtabmap/utilite/include/rtabmap/utilite/ULogger.h:60:23: note: in expansion of macro ‘ULOGGER_INFO’
 #define UINFO(...)    ULOGGER_INFO(__VA_ARGS__)
                       ^
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:66:3: note: in expansion of macro ‘UINFO’
   UINFO("FRAME_WIDTH        %f", _capture.get( CV_CAP_PROP_FRAME_WIDTH ));
   ^
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:67:48: error: ‘CV_CAP_PROP_FRAME_HEIGHT’ was not declared in this scope
   UINFO("FRAME_HEIGHT       %f", _capture.get( CV_CAP_PROP_FRAME_HEIGHT ));
                                                ^
/home/nvidia/rtabmap/utilite/include/rtabmap/utilite/ULogger.h:51:89: note: in definition of macro ‘ULOGGER_LOG’
 #define ULOGGER_LOG(level, ...) ULogger::write(level, __FILE__, __LINE__, __FUNCTION__, __VA_ARGS__)
                                                                                         ^
/home/nvidia/rtabmap/utilite/include/rtabmap/utilite/ULogger.h:60:23: note: in expansion of macro ‘ULOGGER_INFO’
 #define UINFO(...)    ULOGGER_INFO(__VA_ARGS__)
                       ^
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:67:3: note: in expansion of macro ‘UINFO’
   UINFO("FRAME_HEIGHT       %f", _capture.get( CV_CAP_PROP_FRAME_HEIGHT ));
   ^
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenNICV.cpp:68:51: error: ‘CV_CAP_PROP_OPENNI_FRAME_MAX_DEPTH’ was not declared in this scope
   UINFO("FRAME_MAX_DEPTH    %f mm", _capture.get( CV_CAP_PROP_OPENNI_FRAME_MAX_DEPTH ));
                                                   ^
/home/nvidia/rtabmap/utilite/include/rtabmap/utilite/ULogger.h:51:89: note: in definition of macro ‘ULOGGER_LOG’
 #define ULOGGER_LOG(level, ...) ULogger::write(level, __FILE__, __LINE__, __FUNCTION__, __VA_ARGS__)
                                                                                         ^
[... 69 lines cut ...]
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:475: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenNICV.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenNICV.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenni.cpp: In member function ‘void rtabmap::CameraOpenni::image_cb(const boost::shared_ptr<openni_wrapper::Image>&, const boost::shared_ptr<openni_wrapper::DepthImage>&, float)’:
/home/nvidia/rtabmap/corelib/src/camera/CameraOpenni.cpp:89:31: error: ‘CV_RGB2BGR’ was not declared in this scope
  cv::cvtColor(rgbFrame, rgb_, CV_RGB2BGR);
                               ^
/home/nvidia/rtabmap/corelib/src/camera/CameraImages.cpp: In member function ‘virtual rtabmap::SensorData rtabmap::CameraImages::captureImage(rtabmap::CameraInfo*)’:
/home/nvidia/rtabmap/corelib/src/camera/CameraImages.cpp:681:29: error: ‘CV_BGRA2BGR’ was not declared in this scope
      cv::cvtColor(img, out, CV_BGRA2BGR);
                             ^
/home/nvidia/rtabmap/corelib/src/camera/CameraImages.cpp:689:39: error: ‘CV_BayerBG2BGR’ was not declared in this scope
       cv::cvtColor(img, debayeredImg, CV_BayerBG2BGR + _bayerMode);
                                       ^
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:379: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraImages.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraImages.cpp.o] Error 1
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:427: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenni.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenni.cpp.o] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
`

can anyone help me with that?

matlabbe, 2018-10-21:
Are you using OpenCV 4? If so, the commit above should fix those errors.

Bogonek, 2018-10-23:
@matlabbe Thank you for responding!

Yes, I have OpenCV4 in set from JetPack 3.3

After your commit it seems to working, but...

nvidia@tegra-ubuntu:~/rtabmap/build$ make -j4
[  5%] Built target rtabmap_utilite
[  6%] Built target res_tool
[  7%] Built target imagesJoiner
[  8%] Built target extractObject
Scanning dependencies of target rtabmap_core
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RtabmapThread.cpp.o
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Statistics.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriverSqlite3.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBReader.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Recovery.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 12%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Camera.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 12%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 12%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraModel.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 13%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraFreenect.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 13%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraFreenect2.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 13%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraImages.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 14%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraK4W2.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 14%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenni.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 15%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenNI2.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 15%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenNICV.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 15%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRealSense.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 16%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRealSense2.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 16%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRGBDImages.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 16%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoDC1394.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 17%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoFlyCapture2.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 17%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoImages.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 18%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoVideo.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 18%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoZed.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 18%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraVideo.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 19%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/EpipolarGeometry.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 19%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/VisualWord.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[... 366 lines cut ...]
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[... cut ...]
[ 83%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_EditDepthArea.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 84%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/qrc_GuiLib.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 84%] Linking CXX shared library ../../../bin/librtabmap_gui.so
[ 84%] Built target rtabmap_gui
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

matlabbe, 2018-10-23:
For the libGL.so errors:
```
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to drmGetDevices2' /usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to drmCloseOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to drmMap' /usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to drmUnmap' /usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to drmGetDeviceNameFromFd2' /usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to drmGetDevice2'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
```
see step 7 of this guide: https://github.com/introlab/rtabmap_ros#build-from-source-for-nvidia-jetson-with-opencv-4-tegra. You may also check for the libvtkproj4 instructions too.

cheers,
Mathieu

Bogonek, 2018-10-25:
Thank you very much @matlabbe ! 

Now it works :)

Just need to calibrate it, coz it's likely to lost odometry vision in kind of any move...

closed 2018-10-25 by matlabbe