introlab/rtabmap #323: Error while building current upstream master

TSC21, 2018-10-23:
Hi @matlabbe,

While building latest upstream on an Ubuntu 18.04 machine, I face the following errors:
```bash
/home/user/nuno/src/ws/src/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp: In member function 'bool rtabmap::OptimizerG2O::saveGraph(const string&, const std::map<int, rtabmap::Transform>&, const std::multimap<int, rtabmap::Link>&)':
/home/user/nuno/src/ws/src/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1393:49: error: 'PARAM_OFFSET' was not declared in this scope
    fprintf(file, "PARAMS_SE2OFFSET %d 0 0 0\n", PARAM_OFFSET);
                                                 ^~~~~~~~~~~~
/home/user/nuno/src/ws/src/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1393:49: note: suggested alternative: 'ADJ_OFFSET'
    fprintf(file, "PARAMS_SE2OFFSET %d 0 0 0\n", PARAM_OFFSET);
                                                 ^~~~~~~~~~~~
                                                 ADJ_OFFSET
/home/user/nuno/src/ws/src/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1401:5: error: 'PARAM_OFFSET' was not declared in this scope
     PARAM_OFFSET,
     ^~~~~~~~~~~~
/home/user/nuno/src/ws/src/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1401:5: note: suggested alternative: 'ADJ_OFFSET'
     PARAM_OFFSET,
     ^~~~~~~~~~~~
     ADJ_OFFSET
/home/user/nuno/src/ws/src/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1455:26: error: 'PARAM_OFFSET' was not declared in this scope
      to = uFormat(" %d", PARAM_OFFSET);
                          ^~~~~~~~~~~~
/home/user/nuno/src/ws/src/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:1455:26: note: suggested alternative: 'ADJ_OFFSET'
      to = uFormat(" %d", PARAM_OFFSET);
                          ^~~~~~~~~~~~
                          ADJ_OFFSET
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
```

This is being built inside a catkin workspace using `catkin_tools`.

matlabbe, 2018-10-23:
I'll check on docker shortly, but looking at the code, PARAM_OFFSET is defined in the same file [here](https://github.com/introlab/rtabmap/blob/d2abc3a237363a7da9e46716c76714ed939be35e/corelib/src/optimizer/OptimizerG2O.cpp#L69).

matlabbe, 2018-10-23:
It should build now with the last commit above,
cheers

closed 2018-10-23 by matlabbe