introlab/rtabmap #325: ZED sdk greyed out

TheBricktop, 2018-10-29:
Hello again,
Im trying out new zed mini stereo camera and I want to test it inside RTAB map but with zed sdk.
Is there any way to get RTABmap build with ZED sdk?
Cheers.

Myzhar, 2018-10-30:
Have you seen the ZED example using ROS?
https://github.com/stereolabs/zed-ros-wrapper/tree/master/examples/zed_rtabmap_example

TheBricktop, 2018-10-30:
Is it going to work on Win10?

Myzhar, 2018-10-30:
Currently ZED wrapper for ROS only supports Ubuntu since Windows 10 porting of ROS is not yet fully tested

TheBricktop, 2018-10-31:
Thank You very much.
I'll try the earlier build

Rafcin, 2019-02-04:
Hi, I would just like to reopen this, I rebuilt RTABMAP with the SDK sdk and it still seems to be greyed out

natowi, 2020-06-16:
> latest binary release on Windows with ZED sdk is 0.16.3.

"RTABMap-0.16.3-win64-ZED21-CUDA80.exe"

rtabmap 0.16.3 detects the sdk but not the ZED 2 camera
rtabmap 0.20 - zed sdk greyed out

Is a specific ZED sdk cuda version required or something like this?
Or is the zed sdk excluded from the appveyor build?

matlabbe, 2020-07-03:
I updated the releases, the latest one has support for ZED sdk 3: https://github.com/introlab/rtabmap/releases/tag/0.20.2

I don't have many windows computers to test on, please let me know if there are problems starting the application (like missing a dll or something)

natowi, 2020-07-03:
Thank you for the update. I installed RTABMap-0.20.2-win64-CUDA10.exe  and get a "Kinect20.dll missing" error message.
Kinect for Windows SDK 2.0 is required to start RTABMap. It might be good to emphasize this in the release notes some more.
I think this issue can be closed now.

matlabbe, 2020-07-03:
Great, I was anticipating this error. I'll see if I could package that dll without causing errors on systems without kinect SDK 2 installed. Till then, I updated the release notes.

closed 2020-07-03 by matlabbe

matlabbe, 2020-07-03:
Follow up: I updated the binaries with the missing dll.