introlab/rtabmap #332: Export error poses

zoumaguanxin, 2018-11-07:
Hi,
    My robot equiped with a wheel odometer, a lidar and a depth camera.  My parameter "frame_id" is set as "base_link". After mapping, I used the `rtabmap` to export poses of `robot`  as RGBD-SLAM.txt format. In general, I should get the robot poses in map frame. But I depicted the exported poses in my occupancy map, the result is obviously incorrect. I found that in your `graph.cpp` you transform the original poses to rgbd-slam word frame and remove the optical  rotation. The following is your code.
```
		Transform t( 0, 0, 1, 0,
				0, -1, 0, 0,
				1, 0, 0, 0);
	        Transform pose = t.inverse() * iter->second;
		t = Transform( 0, 0, 1, 0,
				       -1, 0, 0, 0,
				       0,-1, 0, 0);
		pose = t.inverse() * pose * t;
```
    When my frame_id is base_link, removing the optical roation is very difficult to understandSo I removed the transforms above. I get correct positions of robot but incorrect orientations. The following my code.
```
	T0<< 0, 0, 1, 0,
		0, -1, 0, 0,
		1, 0, 0, 0,
		0, 0, 0, 1;
	T1<<0, 0, 1, 0,
		-1, 0, 0, 0,
		0,-1, 0, 0,
		0, 0, 0, 1;
	pose=T0.inverse()*T1*Pose*T1.inverse();
```

[image]
The orientations of poses in the picture above are incorrect. So whether did I understand something wrongly?  If exporting the 'robot' poses, are they the poses of the $(arg frame_id) in map frame? If there are not "base_link" frame, what does `robot` mean?  Always is it rgb_optical_frame?

matlabbe, 2018-11-08:
The export of rgbd-slam format can be not obvious. The poses are transformed in the coordinates of the motion capture system used in [TUM's RGB-D datasets](https://vision.in.tum.de/data/datasets/rgbd-dataset/file_formats):

> tx ty tz (3 floats) give the position of the optical center of the color camera with respect to the world origin as defined by the motion capture system.
> qx qy qz qw (4 floats) give the orientation of the optical center of the color camera in form of a unit quaternion with respect to the world origin as defined by the motion capture system.

Not that the RAW format (12 values) doesn't have any rotation, they are exported in ROS coordinates. I'll take a look to add a new export format to have also tx ty tz qx qy qz qw but in original ROS coordinate frames.

matlabbe, 2018-11-08:
In the commit above, I added export of RGBD-SLAM format without coordinate transformation. The option has been added to Export menu in main GUI and database viewer.

zoumaguanxin, 2018-12-28:
It works. Thanks very much.