introlab/rtabmap #334: build error with opencv3.4+cuda9.2

WuYingzheng, 2018-11-11:
hi, There was a error when I compile the project with cuda(9.2) enabled opencv(3.4). could anyone help figure this out?

In file included from /usr/local/include/opencv2/core.hpp:56:0,
                 from /usr/local/include/opencv2/highgui.hpp:46,
                 from /usr/local/include/opencv2/highgui/highgui.hpp:48,
                 from /home/wyz/programs/rtabmap-0.18.0/corelib/src/../include/rtabmap/core/VWDictionary.h:32,
                 from /home/wyz/programs/rtabmap-0.18.0/corelib/src/VWDictionary.cpp:28:
/usr/local/include/opencv2/core/traits.hpp: In instantiation of ‘struct cv::traits::Type<cv::DMatch>’:
/usr/local/include/opencv2/core/mat.inl.hpp:223:39:   required from ‘cv::_OutputArray::_OutputArray(std::vector<std::vector<_Tp> >&) [with _Tp = cv::DMatch]’
/home/wyz/programs/rtabmap-0.18.0/corelib/src/VWDictionary.cpp:1089:81:   required from here
/usr/local/include/opencv2/core/traits.hpp:366:8: error: ‘type’ is not a member of ‘cv::DataType<cv::DMatch>’
 { enum { value = DataType<T>::type }; };
        ^
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:787: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/VWDictionary.cpp.o' failed

WuYingzheng, 2018-11-11:
[image]
error just like the image show.

matlabbe, 2018-11-11:
Make sure to clear the build directory of rtabmap and redo cmake to make sure the right opencv include dirs are used. It looks like that there is a mix of different opencv versions of header files. To be more precise, what is the patch version of opencv 3.4.#?

For example, OpenCV 3.4.3 with CUDA 10 has been used for the windows binaries of version [0.18.0](https://github.com/introlab/rtabmap/releases/tag/0.18.0).

WuYingzheng, 2018-11-12:
> Make sure to clear the build directory of rtabmap and redo cmake to make sure the right opencv include dirs are used. It looks like that there is a mix of different opencv versions of header files. [...]

Hi, matlabbe.
Thanks for your reply! I am using opencv3.4 and opencv_contrib3.4, which is cloned from github.  And cuda9.2 was was involved. 
Yesterday, I have solved the problem by using "OPENCV_TRAITS_ENABLE_DEPRECATED" enabled in opencv. Then problem is missing after the re-make re-install operation.
[image]
Take a look at the printscreen.  Something just like this. 
thanks again!

closed 2018-11-12 by WuYingzheng

jakalaka, 2018-12-27:
same problem ...
change 
//#define OPENCV_TRAITS_ENABLE_DEPRECATED
to 
#define OPENCV_TRAITS_ENABLE_DEPRECATED
in modules/core/include/opencv2/core/traits.hpp solve it .

gkouros, 2019-01-23:
> same problem ...
> change
> //#define OPENCV_TRAITS_ENABLE_DEPRECATED
> to
> #define OPENCV_TRAITS_ENABLE_DEPRECATED
> in modules/core/include/opencv2/core/traits.hpp solve it .

I have the same issue with OpenCV4, CUDA10 and rtabmap on master branch (f50018776fd4d796be524d51be4ef0173fc28345), but your suggestion causes opencv compilation to fail with the following errors. Any suggestions?

[image]