introlab/rtabmap #337: Possible error logic?

seanlis, 2018-11-13:
Hi,

https://github.com/introlab/rtabmap/blob/16309e6d1846097cccdf87224330dd8f80bb5de6/corelib/src/util2d.cpp#L206
Should this line be `localMinDisparity -= maxCol-leftPyramid[level].cols+1; `?
If localMinDisparity is added, `center.x+localMinDisparity+halfWin.width+1` would get an even larger value, further crossing the Matrix boundary.

matlabbe, 2018-11-13:
The logic seems okay. 

* `localMinDisparity` can be zero or a negative value
* `maxCol` is at most equal to `leftPyramid[level].cols`

Example:
If `localMinDisparity=0`, `halfWin.width=2` (win size of 5), image size is 640 with largest `center.x` at 637:
https://github.com/introlab/rtabmap/blob/16309e6d1846097cccdf87224330dd8f80bb5de6/corelib/src/util2d.cpp#L203-L207
```
maxCol=637 + 0 + 2 + 1 = 640
localMinDisparity += 640-640-1 = -1
```
Because of the following, `center.x` cannot be larger than 637 in this case:
https://github.com/introlab/rtabmap/blob/16309e6d1846097cccdf87224330dd8f80bb5de6/corelib/src/util2d.cpp#L191

While doing some examples, I [updated](https://github.com/introlab/rtabmap/commit/9efd7c14fc5c81cd9597fb8e724e7261ba8b3b0a) the code to be more robust to bad matches.

seanlis, 2018-11-14:
Yeah, you're right. 
Thanks for the clarification.

closed 2018-11-14 by seanlis