introlab/rtabmap #339: compile failed when set CMAKE_BUILD_TYPE to Debug

Maxinjun, 2018-11-20:
Hi,
I want to debug the lrtabmap, but it occur a issue when I set  CMAKE_BUILD_TYPE to Debug, log as following,
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP
-- Found OpenCV: /usr/local/include;/usr/local/include/opencv
-- Found PCL: /usr/include/pcl-1.7;/usr/local/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.2;/usr/include/x86_64-linux-gnu;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/jsoncpp;/usr/include/hdf5/openmpi;/usr/include/libxml2;/usr/include/python2.7;/usr/include/tcl
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL
-- Found freenect2: /home/maxj/freenect2/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
CMake Warning at cmake_modules/FindG2O.cmake:122 (MESSAGE):
  Latest g2o version detected with c++11 interface (config file:
  /usr/local/include/g2o/config.h).  Make sure g2o is built with
  "-DBUILD_WITH_MARCH_NATIVE=OFF" to avoid segmentation faults caused by
  Eigen.
Call Stack (most recent call first):
  CMakeLists.txt:349 (FIND_PACKAGE)

-- Found g2o: /usr/local/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/eigen3
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Found cvsba: /usr/local/include
-- Found octomap 1.9.0: /usr/local/include
-- Found Pthreads
========OCTOMAP_FOUND===================
-- --------------------------------------------
-- Info :
--   Version : 0.18.1
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Debug
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   PCL_DEFINITIONS = -DEIGEN_USE_NEW_STDVECTOR;-DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET;-DFLANN_STATIC;-Dqh_QHpointer
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With external SQLite3     = YES (License: Public Domain)
--   With Freenect             = NO (libfreenect not found)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = YES (License: Apache v2 and/or GPLv2)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
[... 125 lines cut ...]
[ 53%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_DataRecorder.cpp.o
[ 54%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CameraViewer.cpp.o
[ 54%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CalibrationDialog.cpp.o
[ 54%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ExportDialog.cpp.o
[ 55%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_PostProcessingDialog.cpp.o
[ 55%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ExportCloudsDialog.cpp.o
[ 55%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ExportBundlerDialog.cpp.o
[ 56%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_MapVisibilityWidget.cpp.o
[ 56%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_GraphViewer.cpp.o
[ 57%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CreateSimpleCalibrationDialog.cpp.o
[ 57%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ParametersToolBox.cpp.o
[ 57%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_DepthCalibrationDialog.cpp.o
[ 58%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/3rdParty/moc_QMultiComboBox.cpp.o
[ 58%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_TexturingState.cpp.o
[ 58%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_RecoveryState.cpp.o
[ 59%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_EditDepthArea.cpp.o
[ 59%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/qrc_GuiLib.cpp.o
[ 59%] Linking CXX shared library ../../../bin/librtabmap_guid.so
[ 86%] Built target rtabmap_gui
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

I checked that OCTOMAP_FOUND is true and I add "#include <rtabmap/core/OctoMap.h>" 
but it can not work
BTW it compile successully when CMAKE_BUILD_TYPE is true.

Maxinjun, 2018-11-20:
I uninstall OctoMap and it works fine, I do not get the reason ^^

matlabbe, 2018-11-21:
I reproduced the same error in Debug mode. Instead of uninstalling OctoMap, you could also just uncheck `WITH_OCTOMAP` option with cmake-gui, or add `-DWITH_OCTOMAP=OFF` on cmake command line.

Anyway, I fixed the error in this [commit](https://github.com/introlab/rtabmap/commit/ca27dbd2fe264c0d3a09fb4516f3e704b2820def).

Thx for reporting!
Mathieu

commit ca27dbd2 (2018-11-21): Fixed #339 and #289
closed 2018-11-21 by matlabbe