introlab/rtabmap #342: Disable saving database

nathangeorge1, 2018-12-01:
Is there a way to specify the location of the saved db file? I tried to do 

```$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" database_path:="my/path/to/database"```

as specified in https://github.com/introlab/rtabmap_ros/issues/228 but it still saved it in `~/.ros/rtabmap.db`.

matlabbe, 2018-12-02:
Which version do you have?

Here an example of output:
```bash
$ roslaunch rtabmap_ros rtabmap.launch args:="-d" database_path:="~/my_map.db"

... logging to /home/mathieu/.ros/log/4c855f18-f662-11e8-a238-58b03561ac71/roslaunch-mathieu-MacBookPro-19951.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mathieu-MacBookPro:40481/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/my_map.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 1.0
 * /rtabmap/rtabmap/odom_tf_linear_variance: 1.0
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_normal_k: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmapviz/approx_sync: True
 * /rtabmap/rtabmapviz/frame_id: camera_link
 * /rtabmap/rtabmapviz/odom_frame_id: 
 * /rtabmap/rtabmapviz/queue_size: 10
 * /rtabmap/rtabmapviz/subscribe_depth: True
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/rtabmapviz/subscribe_rgbd: False
 * /rtabmap/rtabmapviz/subscribe_scan: False
 * /rtabmap/rtabmapviz/subscribe_scan_cloud: False
 * /rtabmap/rtabmapviz/subscribe_stereo: False
 * /rtabmap/rtabmapviz/wait_for_transform_duration: 0.2

NODES
  /rtabmap/
    rgbd_odometry (rtabmap_ros/rgbd_odometry)
    rtabmap (rtabmap_ros/rtabmap)
[... 70 lines cut ...]
[ INFO] [1543776272.955133175]: rtabmap: Deleted database "/home/mathieu/my_map.db" (--delete_db_on_start or -d are set).
[ INFO] [1543776272.955207897]: rtabmap: Using database from "/home/mathieu/my_map.db" (0 MB).
[ INFO] [1543776273.313078317]: rtabmap: Database version = "0.17.6".
[ INFO] [1543776273.357599424]: /rtabmap/rtabmap: queue_size    = 10
[ INFO] [1543776273.357678770]: /rtabmap/rtabmap: rgbd_cameras = 1
[ INFO] [1543776273.357725895]: /rtabmap/rtabmap: approx_sync   = true
[ INFO] [1543776273.357984763]: Setup depth callback
[ INFO] [1543776273.470079440]: 
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom,
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info,
   /rtabmap/odom_info
[ INFO] [1543776273.482936509]: rtabmap 0.17.6 started...
```

Note the following lines:
```bash
[ INFO] [1543776272.955133175]: rtabmap: Deleted database "/home/mathieu/my_map.db" (--delete_db_on_start or -d are set).
[ INFO] [1543776272.955207897]: rtabmap: Using database from "/home/mathieu/my_map.db" (0 MB).
```

cheers,
Mathieu

nathangeorge1, 2018-12-04:
hi @matlabbe; not sure but I'll check sometime hopefully this month

nathangeorge1, 2018-12-15:
for versions:
1. rtabmap: 0.17.6
2. ROS: kinetic

I was able to successfully move the location to where the database is stored; but how do I prevent it from being stored in the first place?

matlabbe, 2018-12-18:
Change that [line](https://github.com/introlab/rtabmap_ros/blob/e0c58ebe35e3cbb2f6a2087088fabc3201c6821a/launch/rtabmap.launch#L42) to:
 ```xml
<arg name="database_path" default=""/>
```

You should have warnings like those ones when starting and closing rtabmap.launch (telling that no database is saved):
```bash
[ INFO] [1545162636.791451078]: rtabmap: database_path parameter not set, the map will not be saved.
[ WARN] (2018-12-18 14:50:36.799) Rtabmap.cpp:302::init() Using empty database. Mapping session will not be saved unless it is closed with an output database path.
... 
rtabmap: Saving database/long-term memory... (located at )
[ WARN] (2018-12-18 14:51:05.233) DBDriverSqlite3.cpp:440::disconnectDatabaseQuery() Database was initialized with an empty url (in memory). To save it, the output url should not be empty. The database is thus closed without being saved!
rtabmap: Saving database/long-term memory...done! (located at , 0 MB)
```

cheers,
Mathieu

nathangeorge1, 2019-01-02:
great that worked! This will help because my Jetson TX1 has little disk space

closed 2019-01-02 by nathangeorge1

Ivancaminal72, 2020-11-04:
There is a way to set the parameter without modifying the default.launch file? I've tried with setting database_path:="" or ! like is mentioned [here](https://answers.ros.org/question/296116/provide-empty-string-as-roslaunch-argument-substitution/) or even /dev/null but seems that the first two are appended to ~.ros/, and the latter as a complete replace, but both causing this similar error:
```
[ INFO] [1604486415.128057096]: rtabmap: Using database from ~/.ros/! (0 MB).
[FATAL] (2020-11-04 11:40:15.128) Rtabmap.cpp:298::init() Condition (UFile::getExtension(_databasePath).compare("db") == 0) not met!
[rtabmap/rtabmap-4] process has died [pid 18918, exit code -11
```