introlab/rtabmap #343: double free or corruption (out) (Ubuntu 18.04 / ROS Melodic)

matlabbe, 2018-12-02:
```bash
$ ./rtabmap
double free or corruption (out)
Aborted (core dumped)
```

GDB output:
```bash
gdb ./rtabmap
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./rtabmap...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/inogec/workspace/rtabmap/bin/rtabmap 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffb4710700 (LWP 16206)]
double free or corruption (out)

Thread 1 "rtabmap" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff4ed5e97 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff4ed7801 in __GI_abort () at abort.c:79
#2  0x00007ffff4f20897 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff504db9a "%s\n")
    at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff4f2790a in malloc_printerr (str=str@entry=0x7ffff504f870 "double free or corruption (out)") at malloc.c:5350
#4  0x00007ffff4f2ee75 in _int_free (have_lock=0, p=0x555555d12390, av=0x7ffff5282c40 <main_arena>) at malloc.c:4278
#5  0x00007ffff4f2ee75 in __GI___libc_free (mem=0x555555d123a0) at malloc.c:3124
#6  0x00007fffeb8bba6d in g2o::EdgeProjectP2MC_Intrinsics::~EdgeProjectP2MC_Intrinsics() ()
    at /opt/ros/melodic/lib/libg2o_types_sba.so
#7  0x00007fffeb8b553e in  () at /opt/ros/melodic/lib/libg2o_types_sba.so
#8  0x00007ffff7de5733 in call_init (env=0x7fffffffdda8, argv=0x7fffffffdd98, argc=1, l=<optimized out>) at dl-init.c:72
#9  0x00007ffff7de5733 in _dl_init (main_map=0x7ffff7ffe170, argc=1, argv=0x7fffffffdd98, env=0x7fffffffdda8) at dl-init.c:119
#10 0x00007ffff7dd60ca in _dl_start_user () at /lib64/ld-linux-x86-64.so.2
#11 0x0000000000000001 in  ()
#12 0x00007fffffffe122 in  ()
#13 0x0000000000000000 in  ()
```

matlabbe, 2018-12-02:
If we look at the cmake of rtabmap, on this system PCL is built with `-march=native` compile option:
```bash
$ cd rtabmap/build
$ cmake ..
...
--   PCL_DEFINITIONS = -DDISABLE_OPENNI2;-DDISABLE_PCAP;-DDISABLE_PNG;-DDISABLE_LIBUSB_1_0
--   PCL_COMPILE_OPTIONS = -march=native;-msse4.2;-mfpmath=sse
...
```

We have to rebuilt g2o from source with BUILD_WITH_MARCH_NATIVE option enabled:
```
$ sudo apt remove ros-melodic-libg2o
$ git clone https://github.com/RainerKuemmerle/g2o.git
$ cd g2o
$ git checkout 20170730_git
$ mkdir build
$ cd build
$ cmake -DBUILD_WITH_MARCH_NATIVE=ON ..
$ make -j4
$ sudo make install
```

Then rebuild rtabmap. Note that this approach can apply to other dependencies requiring EIgen like libpointmatcher (with libnabo) and gtsam. They should all be built with `-march=native` if PCL has been built with this compile option.

closed 2018-12-02 by matlabbe

likairanran, 2021-11-29:
Excuse me ,I want  to see the real reason for this "double free "issue .

likairanran, 2021-11-29:
Does it have sth to do with "memory alignment",I wonder more,Thanks.

matlabbe, 2021-11-29:
This post https://github.com/David-Willo/eigen_catkin/wiki/Eigen-Memory-Issues explains many pitfalls when using eigen and memory alignment issues.

an99990, 2023-04-09:
currently have this issue with rtbamap 0.21 and master branch of g2o

matlabbe, 2023-04-10:
Just tried with latest g2o and cannot reproduce this error. Try launching in gdb and show backtrace.
```
$ gdb rtabmap
#wait for crash, then type bt
$ bt
```

an99990, 2023-04-10:
[image]

an99990, 2023-04-10:
solution was to built PCL again, and gtsam using eigen system version
`cmake -DGTSAM_USE_SYSTEM_EIGEN=ON`