introlab/rtabmap #350: How dosed RTABMap reject loop closure actually?

zoumaguanxin, 2019-01-09:
Hi,

In your paper(Apperance-Based Loop Closure), the highest loop closure hypothesis is accepted if the new location hypothesis is lower than the loop location threshold. But in the following code, https://github.com/introlab/rtabmap/blob/2a8e5be3612922b254000ae624fb669aed455965/corelib/src/Rtabmap.cpp#L1623
if( _highestHypothesis.second < _loopRatio*lastHighestHypothesis.second), RTABMap rejects it? I can't understand it because the highestHypothesis seems to be the lowest probability of  the new location.

regards
Shawn

matlabbe, 2019-01-09:
Hi Shawn,

In contrast to [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/b/bc/TRO2013.pdf) (Section III-D), `_highestHypothesis.second` is the sum of all loop closure hypotheses (excluding new location):
https://github.com/introlab/rtabmap/blob/2a8e5be3612922b254000ae624fb669aed455965/corelib/src/Rtabmap.cpp#L1587-L1588
so the comparison to threshold is if `_highestHypothesis.second` is over loop closure threshold (instead of if new place hypothesis is under threshold):
https://github.com/introlab/rtabmap/blob/2a8e5be3612922b254000ae624fb669aed455965/corelib/src/Rtabmap.cpp#L1611

The result is the same as in the paper, it is just how it is written which is different.

The code you are referring are about another mechanism to reject loop closures that is not discussed in the paper. By default it is ignored as parameter `_loopRatio=0` by [default](https://github.com/introlab/rtabmap/blob/2a8e5be3612922b254000ae624fb669aed455965/corelib/include/rtabmap/core/Parameters.h#L195). 

cheers,
Mathieu

zoumaguanxin, 2019-01-10:
In your code, the default value of Rtabmap/LoopThr still is 0.11, is that fine?

matlabbe, 2019-01-10:
For a more robust visual relocation, you can increase the minimum of inliers required to accept a hypothesis. The parameter is `Vis/MinInliers` (default 20).

You could theoretically increase `Rtabmap/LoopThr` to avoid bad loop closure hypotheses, however in practice I've found better to tune `Vis/MinInliers` instead, as increasing `Rtabmap/LoopThr` would make the detector discarding many true positives.

You can refine (maybe reject) loop closures with laser scans. You have to set `Reg/Strategy` to 1.

zoumaguanxin, 2019-01-14:
Thanks for your advice.