introlab/rtabmap #351: ld: library not found for -lflann clang: error: linker command failed with exit code 1

koji, 2019-01-11:
Does anyone have any clues to solve the following error?

```shell
[ 40%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/treeoptimizer2.cpp.o
1 warning generated.
[ 40%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/edge_se2Switchable.cpp.o
In file included from /Users/koji.kanao/Desktop/Prj_repo/rtabmap-0.18.0/corelib/src/optimizer/toro3d/treeoptimizer2.cpp:44:
In file included from /Users/koji.kanao/Desktop/Prj_repo/rtabmap-0.18.0/corelib/src/optimizer/toro3d/treeoptimizer2.hh:47:
In file included from /Users/koji.kanao/Desktop/Prj_repo/rtabmap-0.18.0/corelib/src/optimizer/toro3d/posegraph2.hh:46:
In file included from /Users/koji.kanao/Desktop/Prj_repo/rtabmap-0.18.0/corelib/src/optimizer/toro3d/posegraph.hh:49:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/set:389:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__tree:1819:22: warning:
      the specified comparator type does not provide a const call operator [-Wuser-defined-warnings]
                     __trigger_diagnostics()), "");
                     ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/set:809:28: note:
      in instantiation of member function
      'std::__1::__tree<AISNavigation::TreePoseGraph<AISNavigation::Operations2D<double> >::Edge *,
      AISNavigation::EVComparator<AISNavigation::TreePoseGraph<AISNavigation::Operations2D<double> >::Edge *>,
      std::__1::allocator<AISNavigation::TreePoseGraph<AISNavigation::Operations2D<double> >::Edge *>
      >::~__tree' requested here
class _LIBCPP_TEMPLATE_VIS multiset
                           ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__tree:970:7: note:
      from 'diagnose_if' attribute on '__trigger_diagnostics':
      _LIBCPP_DIAGNOSE_WARNING(!__invokable<_Compare const&, _Tp const&, _Tp const&>::value,
      ^                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__config:1130:20: note:
      expanded from macro '_LIBCPP_DIAGNOSE_WARNING'
    __attribute__((diagnose_if(__VA_ARGS__, "warning")))
                   ^           ~~~~~~~~~~~
1 warning generated.
[ 40%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/edge_se3Switchable.cpp.o
1 warning generated.
[ 41%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/edge_switchPrior.cpp.o
[ 41%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/types_g2o_robust.cpp.o
[ 42%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/vertex_switchLinear.cpp.o
[ 42%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OctoMap.cpp.o
[ 42%] Linking CXX shared library ../../../bin/librtabmap_core.dylib
ld: library not found for -lflann
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [../bin/librtabmap_core.0.18.0.dylib] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
```

When I ran `cmake ..`, actually terminal showed the below.
I installed FLANN by Homebrew.

```shell
-- Eigen found (include: /usr/local/include/eigen3, version: 3.3.7)
-- FLANN found (include: /usr/local/Cellar/flann/1.9.1_6/include, lib: flann;flann_cpp)
-- FLANN found (include: /usr/local/Cellar/flann/1.9.1_6/include, lib: flann;flann_cpp)
-- QHULL found (include: /usr/local/include, lib: optimized;/usr/local/lib/libqhull_p.dylib;debug;/usr/local/lib/libqhull_p.dylib)
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /Library/Frameworks/Mono.framework/Headers /usr/lib/libsqlite3.dylib
-- Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES)
-- Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES)
-- Could NOT find OpenMP (missing: OpenMP_C_FOUND OpenMP_CXX_FOUND)
-- Not found OpenMP
-- Found OpenCV: /usr/local/include;/usr/local/include/opencv
-- Found PCL: /usr/local/Cellar/pcl/1.9.1/include/pcl-1.9;/usr/local/include/eigen3;/usr/local/include;/usr/local/Cellar/vtk/8.1.2/include/vtk-8.1;/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include;/usr/local/opt/python@2/Frameworks/Python.framework/Versions/2.7/include/python2.7;/usr/local/opt/hdf5/include;/usr/local/opt/szip/include;/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/libxml2;/usr/local/Cellar/flann/1.9.1_6/include
```

Thanks!

matlabbe, 2019-01-11:
FLANN library flag "`-lflann`" is set by the PCLConfig.cmake file [here](https://github.com/PointCloudLibrary/pcl/blob/f3cc2789ee4108c990e39b7e3d9d2932b08cbd5b/PCLConfig.cmake.in#L650). You may try to add flann library dir to DYLD_LIBRARY_PATH or that PCLConfig.cmake uses to absolute path of the library.

koji, 2019-01-12:
@matlabbe thank you for your reply.

I checked `ccmake` toggle and dylib paths are there. 
```
 pkgcfg_lib_FLANN_flann           /usr/local/Cellar/flann/1.9.1_6/lib/libflann.dylib
 pkgcfg_lib_FLANN_flann_cpp       /usr/local/Cellar/flann/1.9.1_6/lib/libflann_cpp.dylib
```
Still get the same error. Do you have any ideas to solve this issue?

Thank you for your help!

matlabbe, 2019-01-14:
On my mac, I noticed that I may have modified the installed `/usr/local/share/pcl-1.9/Modules/FindFLANN.cmake` to make it work to (note how the pkgconfig part is commented):
```cmake
###############################################################################
# Find FLANN
#
# This sets the following variables:
# FLANN_FOUND - True if FLANN was found.
# FLANN_INCLUDE_DIRS - Directories containing the FLANN include files.
# FLANN_LIBRARIES - Libraries needed to use FLANN.
# FLANN_DEFINITIONS - Compiler flags for FLANN.
# If FLANN_USE_STATIC is specified and then look for static libraries ONLY else
# look for shared ones

if(FLANN_USE_STATIC)
  set(FLANN_RELEASE_NAME flann_cpp_s)
  set(FLANN_DEBUG_NAME flann_cpp_s-gd)
else(FLANN_USE_STATIC)
  set(FLANN_RELEASE_NAME flann_cpp)
  set(FLANN_DEBUG_NAME flann_cpp-gd)
endif(FLANN_USE_STATIC)

#find_package(PkgConfig QUIET)
#if (FLANN_FIND_VERSION)
#    pkg_check_modules(FLANN flann>=${FLANN_FIND_VERSION})
#else(FLANN_FIND_VERSION)
#    pkg_check_modules(FLANN flann)
#endif(FLANN_FIND_VERSION)

#if(NOT FLANN_FOUND)
    find_path(FLANN_INCLUDE_DIR flann/flann.hpp
              HINTS "${FLANN_ROOT}" "$ENV{FLANN_ROOT}"
              PATHS "$ENV{PROGRAMFILES}/flann" "$ENV{PROGRAMW6432}/flann"
              PATH_SUFFIXES include)

    find_library(FLANN_LIBRARY
                 NAMES ${FLANN_RELEASE_NAME}
                 HINTS "${FLANN_ROOT}" "$ENV{FLANN_ROOT}"
                 PATHS "$ENV{PROGRAMFILES}/flann" "$ENV{PROGRAMW6432}/flann"
                 PATH_SUFFIXES lib)

    find_library(FLANN_LIBRARY_DEBUG
                 NAMES ${FLANN_DEBUG_NAME} ${FLANN_RELEASE_NAME}
                 HINTS "${FLANN_ROOT}" "$ENV{FLANN_ROOT}"
                 PATHS "$ENV{PROGRAMFILES}/flann" "$ENV{PROGRAMW6432}/flann"
                 PATH_SUFFIXES lib debug/lib)

    if(NOT FLANN_LIBRARY_DEBUG)
      set(FLANN_LIBRARY_DEBUG ${FLANN_LIBRARY})
    endif(NOT FLANN_LIBRARY_DEBUG)

    set(FLANN_INCLUDE_DIRS ${FLANN_INCLUDE_DIR})
    set(FLANN_LIBRARIES optimized ${FLANN_LIBRARY} debug ${FLANN_LIBRARY_DEBUG})

    include(FindPackageHandleStandardArgs)
    find_package_handle_standard_args(FLANN DEFAULT_MSG FLANN_LIBRARY FLANN_INCLUDE_DIR)

    mark_as_advanced(FLANN_LIBRARY FLANN_LIBRARY_DEBUG FLANN_INCLUDE_DIR)
#endif()

if(FLANN_FOUND)
  message(STATUS "FLANN found (include: ${FLANN_INCLUDE_DIRS}, lib: ${FLANN_LIBRARIES})")
  if(FLANN_USE_STATIC)
    add_definitions(-DFLANN_STATIC)
  endif(FLANN_USE_STATIC)
endif(FLANN_FOUND)

```
You may try with this one. When doing cmake in rtabmap afterwards, I can see the full paths of the flann library:
```
-- FLANN found (include: /opt/local/include, lib: optimized;/opt/local/lib/libflann_cpp.dylib;debug;/opt/local/lib/libflann_cpp.dylib)
```

cheers,
Mathieu