introlab/rtabmap #358: Sending new sections of 3D Map as it is built through a topic.

bto-ealdaz, 2019-02-05:
First of all thank you for sharing such an amazing tool, I've been blown away by it's easy of use and performance.

I have been using it in ROS mode, and I was wondering if you could give me any pointers on whether it would be possible to modify RTABMap to send the Octomap or (point cloud topic /cloud_map)not as a whole map as it is the case now, but in increments. That is: as new voxels are added to the octomap they would be sent also thru a topic.
This assumes that the code at some point (maybe at keyframes? ) updates the octomap incrementally... (as in adds new voxels to it) which might not be the case... Currently digging through the code to try to figure it out, but as I'm finding it a bit hard thought I'd ask for help.

Thanks for your time
Eduardo

PS I must admit to being a bit confused as to the difference between Occupancy Grids / Octomap and Maps, which are terms used thorough the documentation, and seem broadly equivalent to me but I suspect they are not as there are calls in the code to `update` functions defined in files OccupancyGrid.cpp and OctoMap.cpp. If you have any time to clarify this too it would be greatly appreciated.

matlabbe, 2019-02-10:
OctoMap and OccupancyGrid have the same purpose, creating occupancy grid maps in different formats. OccupancyGrid is mostly used for 2D mapping. It can handle 3D local occupancy grids, but the output is a point cloud and there is no ray tracing (only in 2D case). OctoMap would be used for 3D voxel maps, and 3D ray tracing can be enabled (Grid/RayTracing).

In the code, the maps are created incrementally, until a loop closure is found and the whole map is optimized. It is why the whole map is always republished, in case of such events. There is no way to publish only what's new in the map for map/octomap topics. If it is network bandwidth problem where you would only want increments, you would have to subscribe to /rtabmap/mapData topic and reconstruct the map on client side (e.g. like rtabmap/map_assembler node). That way, if a loop closure happens, the map would be re-created on client side without any bandwidth cost.

bto-ealdaz, 2019-02-15:
Thanks very much for your answer, very helpful.
The  /rtabmap/mapData topic is a very useful concept, would i be correct in assuming it essentially republishes what is stored in the database?

matlabbe, 2019-02-20:
Yes, it should contain the same data saved in the database for the latest node added to map.