introlab/rtabmap #359: ULogger - enabling UDEBUG messages to console

bto-ealdaz, 2019-02-05:
I am debugging step by step and would like to have  the UDEBUG messages printed to console.
I have tried running rtabmap with these parameters:
`rtabmap --udebug --logconsole `

Have also modified main.cpp
```	
/* Set logger type */
	ULogger::setType(ULogger::kTypeConsole);
	ULogger::setLevel(ULogger::kDebug);  
```
And even modified ULogger.cpp to these types
```
ULogger::Level ULogger::level_ = kDebug ;//kInfo; // By default, we show all info msgs + upper level (Warning, Error)
ULogger::Type ULogger::type_ = ULogger::kTypeConsole;//kTypeNoLog; // Default nothing
```
Still no UDEBUG messages in console.

Any help appreciated...
Thanks
Eduardo

bto-ealdaz, 2019-02-05:
I've got it to work by setting it inside MainWindow.cpp  applyPrefSettings
No doubt this is the wrong place to do it...

```	if(flags & PreferencesDialog::kPanelLogging)
	{
		UDEBUG("Logging settings changed...");
        ULogger::setType(ULogger::kTypeConsole);
        ULogger::setLevel(ULogger::kDebug);
		//ULogger::setLevel((ULogger::Level)_preferencesDialog->getGeneralLoggerLevel());
		ULogger::setEventLevel((ULogger::Level)_preferencesDialog->getGeneralLoggerEventLevel());
		//ULogger::setType((ULogger::Type)_preferencesDialog->getGeneralLoggerType(),
		//				 (_preferencesDialog->getWorkingDirectory()+QDir::separator()+LOG_FILE_NAME).toStdString(), true);
		ULogger::setPrintTime(_preferencesDialog->getGeneralLoggerPrintTime());
		ULogger::setPrintThreadId(_preferencesDialog->getGeneralLoggerPrintThreadId());
		ULogger::setTreadIdFilter(_preferencesDialog->getGeneralLoggerThreads());
	}```

bto-ealdaz, 2019-02-05:
From the above it was clearly something set in the preferences window of rtabmap...
[image]

closed 2019-02-06 by bto-ealdaz

matlabbe, 2019-02-06:
Indeed, when doing "`$ rtabmap --udebug`", there are debug logs until the MainWindow load the configuration from its config.ini file, which overwrites the option set on command line. We should set it explicitly in the UI preferences as you did.