introlab/rtabmap #361: Generating point clouds in C++

ssemenova, 2019-02-24:
Is there a way to get a generated point cloud in a similar way to the way the GUI generates a point cloud after a database upload, but programmatically and without relying on the GUI? I've looked into the code in ```MainWindow``` and ```ExportCloudsDialogue``` and tried to separate out the logic in those so they don't rely on any GUI features, but I didn't get far in getting this to work.

I know you can do this through ROS by subscribing to a topic, but I'm not using (and in fact, can't use) rtabmap through ROS.

matlabbe, 2019-02-25:
There is an example here: https://github.com/introlab/rtabmap/blob/master/examples/Export/main.cpp
There are not as many options than in the GUI, but the basics are there to create a point cloud, a mesh or a texture mesh of the map.

ssemenova, 2019-02-25:
Ah, yes, I ended up modifying that last night and I think I got it working. I think that's the way to go, thank you!

closed 2019-02-25 by ssemenova