introlab/rtabmap #363: EdgeSE3Prior not declared in scope

ahmadhasan2k8, 2019-02-27:
I just started using RTAB map. I have an installation in a ubuntu 16.04 system where it builds without problem. But I have another computer where I have Ubuntu 18.04 and I get the following build error:

```
In member function ‘virtual std::map<int, rtabmap::Transform> rtabmap::OptimizerG2O::optimize(int, const std::map<int, rtabmap::Transform>&, const std::multimap<int, rtabmap::Link>&, cv::Mat&, std::__cxx11::list<std::map<int, rtabmap::Transform> >*, double*, int*)’:
/home/aumi/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:497:8: error: ‘EdgeSE3XYZPrior’ was not declared in this scope
        EdgeSE3XYZPrior * priorEdge = new EdgeSE3XYZPrior();
        ^~~~~~~~~~~~~~~
/home/aumi/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:497:8: note: suggested alternative:
In file included from /usr/local/include/g2o/types/slam3d/types_slam3d.h:50:0,
                 from /home/aumi/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:61:
/usr/local/include/g2o/types/slam3d/edge_se3_xyzprior.h:39:30: note:   ‘g2o::EdgeSE3XYZPrior’
   class G2O_TYPES_SLAM3D_API EdgeSE3XYZPrior : public BaseUnaryEdge<3, Vector3, VertexSE3>
                              ^~~~~~~~~~~~~~~
/home/aumi/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:497:26: error: ‘priorEdge’ was not declared in this scope
        EdgeSE3XYZPrior * priorEdge = new EdgeSE3XYZPrior();
                          ^~~~~~~~~
/home/aumi/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:497:26: note: suggested alternative: ‘priorsIgnored’
        EdgeSE3XYZPrior * priorEdge = new EdgeSE3XYZPrior();
                          ^~~~~~~~~
                          priorsIgnored
/home/aumi/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:497:42: error: expected type-specifier before ‘EdgeSE3XYZPrior’
        EdgeSE3XYZPrior * priorEdge = new EdgeSE3XYZPrior();
```
To test I changed the line temporarily and made it
```
 g2o::EdgeSE3XYZPrior * priorEdge = new g2o::EdgeSE3XYZPrior();
```
which builds and I can run ./rtabmap just fine it seems like. Could you please give me an idea if it's somehow related to ubuntu 18.04 or something else?

matlabbe, 2019-03-03:
I cannot reproduce this error on melodic ([docker image can be built](https://hub.docker.com/r/introlab3it/rtabmap_ros/tags)). Do you use a g2o built from source, or the binaries coming with melodic?

matlabbe, 2019-03-03:
The problem was when using g2o from source. This is fixed in the commit above.

closed 2019-03-03 by matlabbe

ahmadhasan2k8, 2019-03-04:
Great. Sorry about late response. I was out during weekend. I will test it and let you know. Thanks for looking into it.

ahmadhasan2k8, 2019-03-05:
It builds fine now. Thanks