introlab/rtabmap #364: Bad pointcloud with Kinect & Windows stand-alone

jattiogbe, 2019-02-28:
Hi everyone,

I am new to RTAB-Map which I found to be really great !
I use the stand-alone (version 0.18.0) with Kinect 360 / OpenNI2, on Windows10. It seems to work well at first but after a couple of minutes of acquisition, here is the resulting pointcloud in my office

[image]

As you can see, the room is duplicated on the right... How come ? Do I need to calibrate the device ? Should I post-process the pointcloud ? Is my 2nd hand kinect in bad shape ?

jattiogbe, 2019-03-01:
Hello again,

I think I found the origin of the problem, running a test with camera viewer, I have an obvious misalignment

[image]

It is explained [here](https://github.com/introlab/rtabmap/wiki/Installation#depth-not-registered-to-rgb) but I can't actually find the way to fix it, working with the stand-alone on Windows10- kinect-OpenNI2.

matlabbe, 2019-03-03:
I don't think there is a problem with the sensor, default calibration is not always good with this kind of sensor. There are couple of recommendations for good scanning on this [page](https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lost-odometry-red-screens) under Odometry Lost section. In particular avoid areas lacking of discriminative visual features, make sure most visual features are under 4 meter form the camera, don't move fast, close loop closures often.

If you did the 3-steps calibration, make sure to set the calibration file path in the corresponding box in Preferences->Source so that it is actually used.

cheers,
Mathieu

jattiogbe, 2019-03-03:
Ok thanks for the advice, I'll try again tomorrow.
What about the misalignment problem ? How can I dix it ? 
Merci ;)

matlabbe, 2019-03-04:
For the misalignment, it is easier to debug this under ROS (ubuntu) with this tutorial: http://wiki.ros.org/openni_launch/Tutorials/ExtrinsicCalibration
 
The 3-steps calibration done in rtabmap is very similar, but you may find debug better each step with the ros version.