introlab/rtabmap #367: Save point cloud and trajectory from RGBD-Mapping [standalone version]

ahmadhasan2k8, 2019-03-06:
I have realsense2 driver, so when I run:
```
./rtabmap-rgbd_mapping 8
```
I see the visualizer start and SLAM running. But the code examples/RGBDMapping/main.cpp:
```

	// Save 3D map
	printf("Saving rtabmap_cloud.pcd...\n");
	std::map<int, Signature> nodes;
	std::map<int, Transform> optimizedPoses;
	std::multimap<int, Link> links;
	rtabmap->get3DMap(nodes, optimizedPoses, links, true, true);
	pcl::PointCloud<pcl::PointXYZRGB>::Ptr cloud(new pcl::PointCloud<pcl::PointXYZRGB>);
	for(std::map<int, Transform>::iterator iter=optimizedPoses.begin(); iter!=optimizedPoses.end(); ++iter)
	{
		Signature node = nodes.find(iter->first)->second;

		// uncompress data
		node.sensorData().uncompressData();

		pcl::PointCloud<pcl::PointXYZRGB>::Ptr tmp = util3d::cloudRGBFromSensorData(
				node.sensorData(),
				4,           // image decimation before creating the clouds
				4.0f,        // maximum depth of the cloud
				0.0f);
		pcl::PointCloud<pcl::PointXYZRGB>::Ptr tmpNoNaN(new pcl::PointCloud<pcl::PointXYZRGB>);
		std::vector<int> index;
		pcl::removeNaNFromPointCloud(*tmp, *tmpNoNaN, index);
		if(!tmpNoNaN->empty())
		{
			*cloud += *util3d::transformPointCloud(tmpNoNaN, iter->second); // transform the point cloud to its pose
		}
	}
	if(cloud->size())
	{
		printf("Voxel grid filtering of the assembled cloud (voxel=%f, %d points)\n", 0.01f, (int)cloud->size());
		cloud = util3d::voxelize(cloud, 0.01f);

		printf("Saving rtabmap_cloud.pcd... done! (%d points)\n", (int)cloud->size());
		pcl::io::savePCDFile("rtabmap_cloud.pcd", *cloud);
		//pcl::io::savePLYFile("rtabmap_cloud.ply", *cloud); // to save in PLY format
	}
	else
	{
		printf("Saving rtabmap_cloud.pcd... failed! The cloud is empty.\n");
	}
```
I'm not sure how do I get there. Because before that, the threads need to be killed. Not really sure how to kill those.

ahmadhasan2k8, 2019-03-11:
@matlabbe I think here this is the problem

```
[ WARN] (2019-03-11 15:06:35.063) Rtabmap.cpp:302::init() Using empty database. Mapping session will not be saved unless it is closed with an output database path.
```
Could you please tell me how to fix this?

matlabbe, 2019-03-16:
In that example, just close the Window (using X button on the title bar), it will make Qt exiting its [main loop](https://github.com/introlab/rtabmap/blob/c8db45b2a46d316fdd4eae6376bfc1a490fca533/examples/RGBDMapping/main.cpp#L202) and do the exporting code above.

ahmadhasan2k8, 2019-03-16:
Ow great, I will try that on Monday and get back to you.

ahmadhasan2k8, 2019-03-18:
It works. Thanks

closed 2019-03-18 by ahmadhasan2k8