introlab/rtabmap #368: Export database segmentation fault [standalone version]

ahmadhasan2k8, 2019-03-06:
Running:
```
./rtabmap-export --mesh 5MappingSessions_OnlineConstraint700ms.db
```
produces segmentation fault.

With GDB:
```
Starting program: /home/ahasan/rtabmap/bin/rtabmap-export --mesh 2loops_workspace_3IT.db
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffcc7a9700 (LWP 28229)]
[New Thread 0x7fffcbfa8700 (LWP 28230)]
[New Thread 0x7fffc97a7700 (LWP 28231)]
[New Thread 0x7fffc6fa6700 (LWP 28232)]
[New Thread 0x7fffc47a5700 (LWP 28233)]
[New Thread 0x7fffbffa4700 (LWP 28234)]
[New Thread 0x7fffbd7a3700 (LWP 28235)]
[New Thread 0x7fffbcfa2700 (LWP 28236)]
[New Thread 0x7fffb87a1700 (LWP 28237)]
[New Thread 0x7fffb5fa0700 (LWP 28238)]
[New Thread 0x7fffb379f700 (LWP 28239)]
[New Thread 0x7fffb0d9e700 (LWP 28240)]
[New Thread 0x7fffb059d700 (LWP 28241)]
[New Thread 0x7fffafd9c700 (LWP 28242)]
[New Thread 0x7fffaf59b700 (LWP 28243)]
[New Thread 0x7fffaed9a700 (LWP 28244)]
[New Thread 0x7fffae599700 (LWP 28245)]
[New Thread 0x7fffadd98700 (LWP 28246)]
[New Thread 0x7fffad597700 (LWP 28247)]
[New Thread 0x7fffacd96700 (LWP 28248)]
[New Thread 0x7fffac595700 (LWP 28249)]
[New Thread 0x7fffabd94700 (LWP 28250)]
[New Thread 0x7fffab593700 (LWP 28251)]

Thread 1 "rtabmap-export" received signal SIGSEGV, Segmentation fault.
0x00007ffff093c550 in void Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1, 0, -1, 1> >, Eigen::internal::assign_op<double> >(Eigen::Matrix<double, -1, 1, 0, -1, 1>&, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, Eigen::internal::assign_op<double> const&) () from /usr/local/lib/libg2o_core.so
(gdb) q

```

matlabbe, 2019-03-08:
This is an Eigen error, probably caused by wrong compilation flags in the dependencies. Can you show the output of cmake? This is maybe the cause of #365 #366 #367
```
$ cd rtabmap/build
$ cmake ..
<<copy all output here>> 
```
If PCL or G2O or GTSAM are built from source, depending on the versions, the flag "-march=native" should not be set or not **for all libraries**.

ahmadhasan2k8, 2019-03-08:
cmake output:

```
-- Eigen found (include: /usr/include/eigen3, version: 3.2.92)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- FLANN found (include: , lib: flann_cpp)
-- FLANN found (include: , lib: flann_cpp)
-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
CMake Warning at CMakeLists.txt:202 (MESSAGE):
  PCL compile options contain "-march=native", make sure all libraries using
  Eigen are also compiled with that flag to avoid some segmentation faults
  (with gdb referring to some Eigen functions).

-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP
-- Found OpenCV: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv
-- Found PCL: /usr/local/include/pcl-1.9;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-6.2;/usr/include/x86_64-linux-gnu;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/hdf5/openmpi;/usr/include/jsoncpp;/usr/include/python2.7;/usr/include/libxml2;/usr/include/tcl
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL
CMake Warning at cmake_modules/FindG2O.cmake:122 (MESSAGE):
  Latest g2o version detected with c++11 interface (config file:
  /usr/local/include/g2o/config.h).  Make sure g2o is built with
  "-DBUILD_WITH_MARCH_NATIVE=OFF" to avoid segmentation faults caused by
  Eigen.
Call Stack (most recent call first):
  CMakeLists.txt:339 (FIND_PACKAGE)

-- Found g2o: /usr/local/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/include/eigen3
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Found cvsba: /usr/local/include
-- Found RealSense2: 
-- Found octomap 1.8.1: /opt/ros/kinetic/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.19.0
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DDISABLE_PCAP;-DDISABLE_PNG;-DDISABLE_LIBUSB_1_0
--   PCL_COMPILE_OPTIONS = -march=native;-msse4.2;-mfpmath=sse
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With external SQLite3     = YES (License: Public Domain)
--   With Freenect             = NO (libfreenect not found)
--   With OpenNI2              = NO (OpenNI2 not found)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With dc1394               = NO (dc1394 not found)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--  *With g2o                  = YES (License: BSD)
--  *With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--  *With libpointmatcher      = NO (libpointmatcher not found)
--   With loam_velodyne        = NO (loam_velodyne not found)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = YES (License: Apache-2)
--   With OCTOMAP              = YES (License: BSD)
--   With CPUTSDF              = NO (CPUTSDF not found)
--   With OpenChisel           = NO (open_chisel not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With okvis                = NO (okvis not found)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ahasan/rtabmap/build

```

matlabbe, 2019-03-08:
Can you try with a version without GTSAM?
```
$ cd rtabmap/build
$ cmake -DWITH_GTSAM=OFF ..
$ make 
```
The reason is that g2o and PCL would be built with -march=native (if you didn't change BUILD_WITH_MARCH_NATIVE option when building g2o), and gtsam is by default not built with -march=native.

ahmadhasan2k8, 2019-03-08:
I'll try it now and update you. Thanks

ahmadhasan2k8, 2019-03-08:
So, that solved this issue and #365 #366. 
In that case, do you think if I reinstall GTSAM with -march=native then I will be able to build rtabmap with GTSAM?

Also, #367 is still confusing to me. I'm not sure how to initiate the point cloud saving while running the code on that one. 

Btw, I'll close #365 and #366 now. Will try to build with GTSAM if possible by using -march=native. If that allows me to build with GTSAM I will close this one. 
Thanks

ahmadhasan2k8, 2019-03-11:
I will close the issue here. The solution you suggested works. Thanks. I have made a pull request to [gtsam](https://github.com/borglab/gtsam/pull/3). Hopefully they'll merge, otherwise I'll probably maintain this fork.

closed 2019-03-11 by ahmadhasan2k8

matlabbe, 2019-03-16:
Yes setting -march-native is possible with GTSAM. There could be an explicit cmake option as discussed here: https://bitbucket.org/gtborg/gtsam/issues/435/compile-with-march-native-by-default

On my side, I was enable to enable it just by appending "-march-native" to GTSAM_CMAKE_CXX_FLAGS option.

ahmadhasan2k8, 2019-03-16:
I saw that. Having some access issues creating pull request on their bitbucket though. Hopefully will sort it out so that I don't have to apply a patch every time. Could you please take a look at  #367?