introlab/rtabmap #371: Does RTAB-Map Support VIO?

rhklite, 2019-03-13:
I'm currently trying to use RTAB-Map along with a ZED-M camera, and was wondering if rtabmap currently supports VIO approach?

from the paper “RTAB-Map as an Open-Source Lidar and Visual SLAM Library for Large-Scale and Long-Term Online Operation” (2018) it says that RTAB map does not take IMU as an input. And I haven't found any sample launch file that gives rtabmap an IMU input.

But i do see in the rtabmap.launch file that stereo input takes in imu input, as seen from the snippet below.

```launch
        <!-- Stereo Odometry -->
        <node if="$(arg stereo)" pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" output="$(arg output)" args="$(arg rtabmap_args) $(arg odom_args)" launch-prefix="$(arg launch_prefix)">
          <remap from="left/image_rect"        to="$(arg left_image_topic_relay)"/>
          <remap from="right/image_rect"       to="$(arg right_image_topic_relay)"/>
          <remap from="left/camera_info"       to="$(arg left_camera_info_topic)"/>
          <remap from="right/camera_info"      to="$(arg right_camera_info_topic)"/>
          <remap     if="$(arg rgbd_sync)" from="rgbd_image" to="$(arg rgbd_topic)"/>
          <remap unless="$(arg rgbd_sync)" from="rgbd_image" to="$(arg rgbd_topic)_relay"/>
          <remap from="odom"                   to="$(arg odom_topic)"/>
          <remap from="imu"                    to="$(arg imu_topic)"/>
      
          <param name="frame_id"                    type="string" value="$(arg frame_id)"/>
          <param name="odom_frame_id"               type="string" value="$(arg vo_frame_id)"/>
          <param name="ground_truth_frame_id"       type="string" value="$(arg ground_truth_frame_id)"/>
          <param name="ground_truth_base_frame_id"  type="string" value="$(arg ground_truth_base_frame_id)"/>
          <param name="wait_for_transform_duration" type="double" value="$(arg wait_for_transform)"/>
          <param name="approx_sync"                 type="bool"   value="$(arg approx_sync)"/>
          <param name="config_path"                 type="string" value="$(arg cfg)"/>
          <param name="queue_size"                  type="int"    value="$(arg queue_size)"/>
          <param name="subscribe_rgbd"              type="bool"   value="$(arg subscribe_rgbd)"/>
          <param name="guess_frame_id"              type="string" value="$(arg odom_guess_frame_id)"/>
          <param name="guess_min_translation"       type="double" value="$(arg odom_guess_min_translation)"/>
          <param name="guess_min_rotation"          type="double" value="$(arg odom_guess_min_rotation)"/>
        </node>
```

inuex3, 2019-03-14:
Hello, I faced the same situation, and I don't know rtabmap really takes in imu input.
Instead, if you are okay with sensor fusion with kalman filter, a sample launch file in realsense_ros uses imu for VIO. Check this repo.
https://github.com/intel-ros/realsense/blob/development/realsense2_camera/launch/opensource_tracking.launch

rhklite, 2019-03-14:
thanks! will try this out

matlabbe, 2019-03-16:
With default odometry approach (e.g. F2M), the imu is used only on initialization to start accordingly to gravity. 

For VIO, rtabmap should be built with MSCKF or OKVIS approach (like in the paper). There is an example here for MSCKF on EuRoC dataset (same can be done with OKVIS with `Odom/Strategy=6`): https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/euroc_datasets.launch

cheers,
Mathieu

rhklite, 2019-03-17:
@matlabbe  thank you very much, will be trying this out.

closed 2019-03-17 by rhklite

matlabbe, 2019-03-22:
Note that I updated https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/euroc_datasets.launch with an OKVIS example, also fixed a bug making OKVIS lagging.

cheers,
Mathieu